BelfrySCAD/BOSL2 #1061: Simple, variable single-axis rounding for cuboid.

RAMilewski, 2023-02-28:
**Is your feature request related to a problem? Please describe.**
I frequently find myself needing a cuboid with z-aligned rounding but with different radii around the cuboid.
The simplest one-line solution to the problem I've found is to use a rect_tube with a vanishingly small isize.

`rect_tube(size = [30,30], isize = [0.001, 0.001], rounding = [5, 0, 15, 15], h = 5);`

 An isize of [0,0] throws an error, and my go-to value of [0.001, 0.001] works for my 3d printing case, but feels wrong and certainly doesn't help code readability.  My 3d printing workflow masks the minuscule hole in the center but it's hardly a general solution.

**Describe the solution you'd like**
Enabling rect_tube-style single-axis rounding in cuboid() would make for clear,  simple code.

`cuboid([30,30,5], rounding = [5, 0, 15, 15], edges = "Z"); `
 
**Describe alternatives you've considered**
Applying an edge_mask() to each z-aligned edge is an obvious method, but this requires substantially more code.

**Additional context**
I'm a relative newcomer to BOSL2 and OpenSCAD so I could well have missed something obvious.

adrianVmariano, 2023-02-28:
The method to make this is linear_extrude applied to rect(), which can round each corner independently.  

Not sure that counts as obvious.

revarbat, 2023-03-01:
Wouldn't `prismoid()` do this just fine?

RAMilewski, 2023-03-01:
OH!  Indeed it does!

I hadn't thought to use prismoid() where size1 = size2.  It offers all of the attachment and anchoring flexibility of cuboid() or rect_tube(), much of which the linear-extrude solution lacks.

I think that's the right answer!

closed 2023-03-01 by RAMilewski