BelfrySCAD/BOSL2 #1075: Puzzling Polar Positioning

RAMilewski, 2023-03-10:
In the Shapes2d Tutorial there is this un-numbered example
```
include <BOSL2/std.scad>

circle(d=50)
    position(polar_to_xy(1,60))
        #circle(d=10);
```

The wiki entry for polar_to_xy(r,theta) defines r as the distance from the origin.

It's probably worth pointing out that in this example (at least) the value of r is ignored.   Any value of r appears to yield the same result.   Is this because we're positioning a child?

adrianVmariano, 2023-03-10:
The position module takes an anchor and anchors are just directions.  You're positioning on the circle in that direction.

RAMilewski, 2023-03-10:
OK.  I think I understand, but on first encounter the code is a bit puzzling.

I was going to suggest something like position(60) since there are no other scalar arguments to position.

But the 1 is necessary to keep expressions like  **position(LEFT+polar_to_xy(1,60))** from throwing an error.
I know that expression makes little sense, but in the 3d analog, spherical_to_xyz(),  it becomes important.

_Many thanks for being so supportive as I stumble through this!_

closed 2023-03-10 by RAMilewski