BelfrySCAD/BOSL2 #1077: Anchor ignoed when used as function.

TwoClocks, 2023-03-12:
These two recs are in different places :

```
rect([ 45, 40 ],anchor=TOP+CENTER);
v=rect([ 45, 40 ],anchor=TOP+CENTER);
#polygon(v);
```

Maybe this is as intended, but I was surprised by this. It's unclear in the `anchor` or `path` docs that this is the case.

revarbat, 2023-03-12:
Three things of note here:
1) `TOP+CENTER` is the same as `TOP`, though this is just a fiddly detail.
2) A 2D shape like `rect()` should treat `TOP` like `BACK`, which seems to be happening with `attachable()`, but not with `reorient()`.  This is likely a bug.
3) If you anchor to `BACK` instead of `TOP`, both shapes anchor correctly in the same place.

adrianVmariano, 2023-03-13:
Anchors with a z component are mishandled in rect() when rounding is not used.  Fixed in PR.

TwoClocks, 2023-03-13:
Thanks for taking a look.

BTW: I just started using join_prism in earnest. Amazing stuff. The longer
I spend w/ this library, the more impressed I've become. Thanks for all
your work.

closed 2023-03-14 by revarbat