BelfrySCAD/BOSL2 #1078: Tagging:  Multi-generation example would be helpful

RAMilewski, 2023-03-12:
It's not immediately obvious why the rounding_hole_mask is ignored in this example:

`include <BOSL2/std.scad>

back_half() { 
  diff() 
    cyl(h=5, d = 20, anchor = BOT)
      attach(TOP) recolor("red") tag("remove") cyl(h = 2, d = 12, anchor = TOP)
        attach(TOP) recolor("blue") tag("keep") tube(od = 20, wall = 2, h = 10, anchor = BOT) 
          attach(TOP) tag("remove") rounding_hole_mask(d = 18, rounding = 3);
}`
[image]

adrianVmariano, 2023-03-12:
So in experimenting a bit, I found that basically what seems to be happening is that you're using the "keep" tag which is the tag to "force this to be kept" and it's overriding the "remove" tag below it.  If you stop using "keep" then I think it works as intended.  I'm not sure if there's a way to make this behave better.  (This whole tagging business is incredibly complex and mysterious.  Like I spent weeks fiddling with it to try to get it to work right all the time, and there always seemed to be cases where it did something unexpected.)  

So:

```
back_half() {
diff()
  cyl(h=5, d = 20, anchor = BOT)
    attach(TOP,overlap=-.01) tag("remove") cyl(h = 2, d = 12, anchor = TOP)
      attach(TOP) tag("don't-remove")tube(od = 20, wall = 2, h = 10, anchor = BOT)
        attach(TOP) tag("remove") rounding_hole_mask(d = 16, rounding = 3)
  ;
}
```

I'll have to look into what exactly is happening and see if there is a way to make it behave better.

RAMilewski, 2023-03-13:
OK,  curiouser and curiouser....

tag("keep") ignores the mask,
tag("**_anything other than remove or keep_**") seems to work, 
leaving tag() empty throws a "tag must be a string" error,
and omitting tag() completely results in this:

[image]

So, explicitly mentioning this behavior and offering a keyword for "something other than remove or keep" in the docs might make it all a bit less mysterious...  not that a particularly appropriate word comes to mind.   If keep weren't already in use then "force" for the current "keep" behavior, and "keep" for "something other" would be fairly obvious.    But I think that ship has sailed.   

The current tag() behavior does make me approach diff() with some trepidation.   I'll keep trying and see if knowing about "something other" makes it all easier.

adrianVmariano, 2023-03-13:
Well, tags just set tags.  By default, diff() pays attention to the "remove" and "keep" tags, as noted in the wiki.  Any other tags are treated as the based objects by diff()...unless you invoke diff() with different tags to use, which you can.  That is, it doesn't have to be "remove" and "keep".  

I'm not sure what you mean by "a keyword for something other than remove or keep".  Did you find the wiki unclear when it laid out the categories of things matching "remove", things matching "keep", and everything else?  

tag("keep") in the spot you have it does **NOT** ignore the mask.  It tags the first item with "keep" and its child (the mask) also gets kept.  That is, the child tag of "remove" is being ignored.   The mask **is** included in the model.  (Note, I found the colors distracting.  Also your mask is in the wrong place and of a questionable size, but you can see it below adding a rim to the model.)  

[image]

Tags are inherited, so if you omit the tag then from the tube then it inherits "remove" from its parent and gets removed...which is to say, it never appears.

adrianVmariano, 2023-03-13:
Wait, that previous example was wrong.  I had changed something.   So it does look like with the mask tagged as "remove" it never appears.   

Actually on thinking a bit more, this is what is supposed to happen:   you have got four objects.   The cylinder has no tag.  It's the base object.  You have two objects removed:  the first cylinder and the mask.  Removing the mask does nothing.  Then you have one object kept, the tube.  The tube is kept as you requested!  Kept objects, as the wiki says, are unioned in **after** other objects are removed.

adrianVmariano, 2023-03-13:
Note that it appears above that you suggest that diff() have a third keyword for what is currently "everything else".  That is impossible.  How can I have a keyword that lists everything else?   Suppose it was designed with a "remove", "force" and "keep".    What happens to objects that don't have any of those three tags???   (And in particular, what about the parent if you neglected to tag is as "keep"?)   Basically there always has to be some catch-all for the objects that are untagged or don't have tags matching the special ones.

RAMilewski, 2023-03-13:
Yes, I did have the wrong rounding_hole_mask diameter for that tube with that wall size.  It should have been 16.

After spending about an hour staring at example 12 in diff() I think I'm beginning to see what's going on, and why my suggestion of a third standard tag string makes little sense.   ....although I do think I'll be using tag("base") for things like the tube in my problem example to keep the code a little more readable for simple folk like me.  

 Examples 8 & 13  in diff() just make my head hurt!    

Many thanks for your patience as I try to get my head around this.

closed 2023-03-13 by RAMilewski