BelfrySCAD/BOSL2 #1082: Any chance of adding rounding1 and rounding 2 to Cuboid and/or Prismoid?

RAMilewski, 2023-03-17:
**Is your feature request related to a problem? Please describe.**
I frequently want to smooth the transition from a cube to it's parent object.

**Describe the solution you'd like**
rounding1 and rounding 2 parameters for cuboid() and/or prismoid() that would behave as with cyl().

**Describe alternatives you've considered**
Can be done with fillet and a cyl for each corner, but parameters would be a cleaner solution.

**Example Code**
```
include <BOSL2/std.scad>
$fn = 36;
    corner = 2;
    rounding1 = -1;
    cuboid(10, rounding = corner, edges = "Z", anchor = BOT);
    grid_copies(size = [6,6]) cyl(r = corner, h = 10, rounding1 = rounding1, anchor = BOT);
    rot_copies(n = 4) 
        fwd(5) left(3)  fillet(l = 6, r = -rounding1, spin = 180, orient = RIGHT, anchor = BOT);
```

**Additional context**

[image]

RAMilewski, 2023-03-17:
Although....  Implementing this for prismoid() might be quite difficult given that the "verticalish" edges are only "ish" and so not necessarily at right angles to the x-y plane.

adrianVmariano, 2023-03-18:
I'm a bit hazy on what you're asking for exactly, but I think you can probably already do it with rounded_prism().  Including for prismoid type shapes.  

It sounds like you're saying you want cuboid to have a totally separate rounding system from the existing one that does something different, like different radius roundings in different places?  Or some kind of override on the existing system?

RAMilewski, 2023-03-18:
Apologies for the haze inducing question!   It was inexperience and hazy thinking on my part confusing prismoid() and rounded_prism().

Rounded_prism() does indeed do what I was hoping to accomplish.

RAMilewski, 2023-03-19:
I just tumbled to why I never thought of rounded_prism() for this problem.  It's not in shapes3d.scad, and so is not mentioned in the wiki section with most of the 3d shapes.  Perhaps a "see also..." link in the wiki sidebar shapes3d section would help for those of us who can't hold the whole of BOSL2 in our heads all at once.

adrianVmariano, 2023-03-19:
I'm not sure what you're asking for.  You think there should be a "see also" on the list of files that shows up on the right sidebar?   That doesn't make sense to me.  

Right now prismoid() has a "See also" that points to rounded_prism() and it seems reasonable to add the same for cuboid.

RAMilewski, 2023-03-19:
I guess I'm not entirely sure either.  Certainly adding a link to rounded_prism() in the cuboid description would be a help.   My learning path for BOSL2 has been largely through exploring the sidebar, and then reaching for the tutorials when I get hopelessly befuddled.  

I've been pondering how to take best advantage of a forthcoming feature in an OpenSCAD extension for the VSCode editor. (https://github.com/Antyos/vscode-openscad/issues/38}.  What I would most like to do is link to just the sidebar in the wiki, but after staring at the page source for a bit, it doesn't look like there's an html anchor there.   Currently I'm thinking that I'll try linking to the Table of Contents.  The CSS will push the wiki sidebar to the bottom when VSCode opens it in a narrow side window.

Just out of curiosity, what tool is used to generate the wiki from the source code?

adrianVmariano, 2023-03-19:
Just to be sure, the side bar you're talking about is the list that starts with "Table of Contents" and then has the "List of Files"?  Or is it something else?  Because that sidebar is after all just a list of the files.  I would think the Table of Contents would be more useful as a learning tool.  Assuming that's the sidebar you mean...it is just a list of files...so it can't offer nuanced guidance about where you need to look for something.  

The tutorials are not very complete, I think.  I'll admit I am not very motivated to write tutorials.  

The Wiki is generated by "docsgen", which Revar wrote.  I think it has a separate github.   If there's some simple mod to the wiki (something that's actually possible) which will make things work better with vscode, Revar could likely add it for you.  Be aware, though, that the wiki is written in the github markdown language and there are a lot of things that seem difficult/impossible to do.

RAMilewski, 2023-03-20:
It's premature now to make any changes to the wiki.  Once Antyos get the change in place I'll try linking to different things and see what works.  The more I muddle on it the more sense it makes to link to the TOC.

On the topic of tutorials, I'm thinking about doing classes at our local maker space along the lines of "OpenSCAD for Complete Beginners".  The 3d printing fans there are using Tinkercad, Fusion360, or Blender (!) to design their objects.
BOSL2 simplifies so many things that I can't imagine trying to teach naked OpenSCAD without it.   (The best teachers I've known all say that the best way to learn something well is to teach it).

As I pull together some lesson plans, I'll try to kludge together some videos to use in class.  If I get anything that works I'll put them up on YouTube.  All of my YouTube searches for BOSL2 seem to come up empty (which is just astounding!), so perhaps there's some fertile ground for evangelizing BOSL2.

adrianVmariano, 2023-03-20:
Well, for my own self, I'd say that videos are the absolute last way I'd want to learn OpenSCAD or BOSL2.  Trying to learn programming by video seems ridiculous.   Just show me the code!  I'm kinda anti-video in general, though, so perhaps not typical.  Video is perhaps suitable for learning skills that are fundamentally about how you move things, and even then I'm not always a fan.   In any case, it can't hurt to have more content out there.

adrianVmariano, 2023-03-20:
With regards to the actual issue, I have added a "See also: rounded_prism()" to cuboid.  It was already there on prismoid().  I've also raised the question about whether the See Also lines should appear right after Usage instead of down after the arguments, where I wonder if they get lost.

RAMilewski, 2023-03-20:
I don't learn well from video either!   But lots of folks a half century younger than I seem to learn quite well that way.  At a minimum the videos would be online ads for OpenSCAD with BOSL2.

revarbat, 2023-03-20:
>Just out of curiosity, what tool is used to generate the wiki from the source code?

I wrote a Python library to generate documentation for OpenSCAD at:

https://GitHub.com/revarbat/openscad_docsgen/

Also available via pypi:

```pip install openscad_docsgen```

RAMilewski, 2023-03-20:
I think I've pulled this discussion far enough off-topic that it ought to close.

closed 2023-03-20 by RAMilewski