BelfrySCAD/BOSL2 #1084: See Also lines right after usage lines?

adrianVmariano, 2023-03-20:
It seems like the "See Also" lines are often after the arguments list, which means they are kind of lost, especially if there is a long description and lots of args.  Should they instead appear after the Usage, along with Topics?

RAMilewski, 2023-03-20:
I think they should be at the end of the Description.  I'm also of the opinion that Description ought to be the topmost paragraph(s) for each function.

For newcomers just the name may not be enough to encourage them to scroll through lengthy (i.e. intimidating) Usage sections.

adrianVmariano, 2023-03-20:
I wonder if you're confused.  The "Description" is sometimes 3 screens long.  The Usage is usually one or two lines except in a few cases.  It would be ridiculous to put the Description before the Usage.  The whole point of the Usage is to give a short quick summary of how to use the function.  It wouldn't do its job if it was buried after other stuff.  

Consider this case, for example:

https://github.com/revarbat/BOSL2/wiki/skin.scad#functionmodule-path_sweep

Isn't the See Also kinda lost after the Arguments section?  And wouldn't it be just as bad sandwiched between Arguments and Description?   So putting See Also at the top, before all the long potentially overwhelming stuff makes it easier to notice and easy to find if you want to look for it.

revarbat, 2023-03-20:
The `See Also:` line feels logically (to me) like it should follow the description, but I agree it may get lost after the `Arguments:` tables.

RAMilewski, 2023-03-21:
Sure, I'll take a crack at writing them.  I'll need to refresh my memory on doing a PR.

RAMilewski, 2023-03-22:
> I would argue that module usage and function usage are pretty similar.

I agree, but in the interest of avoiding the **TL;DR** problem, I would suggest just describing the module action in the Synopsis for most entries.

revarbat, 2023-03-22:
In many cases, a synopsis like "Constructs an XXX" covers both the module and function form, since the function form makes the VNF equivalent of the module's geometry.

adrianVmariano, 2023-03-22:
That's basically what I was thinking.  So do you agree that this project makes sense and you'll have support for a Synopsis header?

revarbat, 2023-03-22:
I like the idea of `Synopsis:` lines.  I can tweak docsgen to show the synopsis for functions and modules in the Alphabetic Index.  Another idea that has been bouncing around the back of my mind is representative images for geometry generating modules, so you can see a small torus image before the Usage lines, but after the Synopsis.

adrianVmariano, 2023-03-22:
Ok.  So then RAMilewski is good to go to write Synopsis lines.

adrianVmariano, 2023-03-23:
Sounds like you're in business.  Hold off on threading.scad and skin.scad, as I have changes for those files.  

If you do happen to notice errors or other deficiencies in the docs feel free to fix.

RAMilewski, 2023-03-23:
I tried adding **DefineHeader(Text): Synopsis** to .openscad_docsgen_rc and the synopses now show, but not in the correct location.  

I also relocated See Also in the .scad file, but in most (but not all) cases its location is unchanged.

What controls the order of docsgen blocks in the .md output file?

adrianVmariano, 2023-03-23:
I've never looked at docsgen, but Revar said that some of the headers have their order forced somehow, so it's not just the order they appear in the code.  Revar is working on updating docsgen so it should be ready soon.

RAMilewski, 2023-03-23:
Ok.  Once we know the order I can rework the .scad file to match (just to preserve sanity).   I was thinking that it could be the Synopsis followed by See Also followed by Topics.   That puts all the things adjacent that someone might need to refer to in order to understand the module.    No big deal for anything in transforms.scad (except maybe frame_map() and apply()).  But more important as we get to more advanced modules.

adrianVmariano, 2023-03-24:
If you group the one-line entries at the top and let docsgen produce the desired order that's probably fine.   

A look at the contents of "Attachable Masks" reveals that the section is misnamed.  It doesn't contain any masks.  It's contains mask attachers.  I'm not sure of the best name for that.  The actual masks are the 3d masks, I think.  And the mask attachers can be used to attach them.

revarbat, 2023-03-24:
I've released openscad_docsgen v2.0.34 ( https://github.com/revarbat/openscad_docsgen/releases/tag/v2.0.34 ).
It supports the Synopsis: line, and fixes the sort order of blocks in Module/Function/Constant docs.

revarbat, 2023-03-24:
I've added Synopsis lines to transforms.scad.

RAMilewski, 2023-03-24:
Thanks for the openscad_docsgen update!

adrianVmariano, 2023-03-24:
As you work through this you can also add Topics entries if it seems appropriate.

adrianVmariano, 2023-03-24:
I suspect that there's a Topics index somewhere and that tutorials links wouldn't really make sense in the topics lists.  There may be some other way to add such links.  Perhaps the See Also entry could do it, though that probably requires a docsgen alteration.

RAMilewski, 2023-03-24:
> A look at the contents of "Attachable Masks" reveals that the section is misnamed.

How about changing that to "Mask Attachment"?

adrianVmariano, 2023-03-24:
Sounds reasonable to me.

RAMilewski, 2023-03-26:
In shapes2d.scad both star() and supershape() take an atype parameter with two valid choices "hull" and "intersect".
However there is no explanation of the difference between the two and no relevant examples.

A search of the Attachments Tutorial also came up empty.   Did I miss something somewhere?

adrianVmariano, 2023-03-27:
Make a new issue for this.   I think it's a big topic that needs work for a satisfactory solution.  "intersect" anchoring anchors to the intersection point of a ray in the anchor direction, and "hull" anchoring finds a point on the convex hull of the shape in the anchor direction.

adrianVmariano, 2023-03-28:
I took a brief scan over your PR and you're definitely on the right track.  I have a few quibbles, but basically it's looking good.  

So quibbles: stroke can take a region.  (dashed stroke basically calls stroke, so they do the same thing except for the dashing.)  And turtle3d can return a list of transforms.  For star just say "star shaped region" and skip the "N points".  

It looks like you reverted some doc tweaks I made to prismoid, though not a huge deal.  (I tried to tighten the list of usage options and added some see also entries.)  

Over all, I wonder if you should say "Creates a Fooblat object or returns a path" or something to that effect.  That is, never mentioning that the function form returns a path may hide this behavior from users who are skimming the docs.  For the 3d objects it could say "or returns a VNF" when appropriate.

RAMilewski, 2023-03-28:
Thanks!  That helps.  I'll make another pass through what I've done so far before I do another PR

RAMilewski, 2023-03-28:
It looks like nothing in partitions.scad appears in the topics lists.

adrianVmariano, 2023-03-29:
I don't really have anything to do with the topics.  Go ahead and add topics if it seems appropriate.

revarbat, 2023-03-29:
Topics were another late addition to the documentation, and not everything has been given topics yet.

RAMilewski, 2023-03-29:
I'll keep an eye open for that as I move forward.   I set the things in partitions.scad to Topics: Partitions, Masking

revarbat, 2023-03-29:
Of note, a blank line after a See Also or Topics line will truncate the docs for the function/module right there, which is not what you want.

RAMilewski, 2023-03-29:
OUCH!   Sorry!   I'll fix that going forward.

RAMilewski, 2023-03-29:
OK... my testing shows it's just completely blank links.
Empty comment lines seem to work ok.

revarbat, 2023-03-30:
Here's a checklist of files and whether or not they have Topics, See Alsos, and Synopses.  I'm starting it blank, and we can check off each file that is verified:

- [ ] affine.scad
- [ ] attachments.scad
- [ ] ball\_bearings.scad
- [ ] beziers.scad
- [ ] bottlecaps.scad
- [ ] builtins.scad
- [ ] color.scad
- [ ] comparisons.scad
- [ ] constants.scad
- [ ] coords.scad
- [ ] cubetruss.scad
- [ ] distributors.scad
- [ ] drawing.scad
- [ ] fnliterals.scad
- [ ] gears.scad
- [ ] geometry.scad
- [ ] hinges.scad
- [ ] joiners.scad
- [ ] linalg.scad
- [ ] linear\_bearings.scad
- [ ] lists.scad
- [ ] masks2d.scad
- [ ] masks3d.scad
- [ ] math.scad
- [ ] metric\_screws.scad
- [ ] modular\_hose.scad
- [ ] mutators.scad
- [ ] nema\_steppers.scad
- [ ] partitions.scad
- [ ] paths.scad
- [ ] polyhedra.scad
- [ ] regions.scad
- [ ] rounding.scad
- [ ] screw\_drive.scad
- [ ] screws.scad
- [ ] shapes2d.scad
- [ ] shapes3d.scad
- [ ] skin.scad
- [ ] sliders.scad
- [ ] strings.scad
- [ ] structs.scad
- [ ] threading.scad
- [ ] transforms.scad
- [ ] trigonometry.scad
- [ ] tripod\_mounts.scad
- [ ] turtle3d.scad
- [ ] utility.scad
- [ ] vectors.scad
- [ ] version.scad
- [ ] vnf.scad
- [ ] walls.scad
- [ ] wiring.scad

revarbat, 2023-03-31:
I'm targeting updating files not in std.scad for the near future.

RAMilewski, 2023-03-31:
As usual, I'm confused.  Does that mean you'd like be to leave things in std.scad to you, or start with those myself going forward.

revarbat, 2023-03-31:
The latter. I plan to work on the docs of files not listed in std.scad. At least until we get nearer to complete.

adrianVmariano, 2023-04-01:
Uh oh, someone did threading.scad, which has massive changes....

adrianVmariano, 2023-04-01:
Maybe merging will be not to hard since I haven't updated the usage messages.

RAMilewski, 2023-04-04:
How do Tutorials make it into the wiki?  I tried using openscad-docsgen on another project and everything seems to happen as expected except the tutorials.  Did I miss a command line argument somewhere?

revarbat, 2023-04-04:
The openscad_docsgen package also installs the script `openscad-mdimggen` which you can run with no arguments to generate the tutorials.

revarbat, 2023-04-06:
Okay, I've run out of std.scad files.  My next target will be vectors.scad, and I will move forward through the alphabet from there.

adrianVmariano, 2023-04-08:
Is there a better way to convey the information other than "or returns a VNF" on the synopsis lines?  I didn't write that on the entries for rounding.scad, and to fit it in and keep the synopses to one line I have to cut out useful description.   It does seem useful, though, to have it clear which ones can return a VNF and which don't.   Could we have some shorthand symbol for return type in the synopsis?  `[GEOM] [VNF]`  The user could hover over it to get elaboration, e.g. "Returns geometry" or "Returns a VNF".  Maybe some even more useful mouseover text could be crafted.

RAMilewski, 2023-04-08:
I like the idea of using rollover text.  I'll have to look up how to make that work in Github Markdown though.

adrianVmariano, 2023-04-08:
Revar has done it elsewhere.  We'd need support for this to happen automatically anyway, or writing the synposis lines will be too painful.

RAMilewski, 2023-04-08:
Maybe not too hard .  Stack Overflow says: 

This is a [hover text](## "your hover text") example.

RAMilewski, 2023-04-08:
Should be able to just do a search and replace to fix "or returns a VNF."

RAMilewski, 2023-04-08:
> We'd need support for this to happen automatically anyway, or writing the synposis lines will be too painful.

Why do you think we'd need support?

adrianVmariano, 2023-04-08:
I think that you're being short-sighted and not thinking about the long term maintenance and big picture here.  If I have to explicitly write out a hover text or 2 every time I write a synopsis line, that's a PITA.   How do you ensure that the hover texts are all the same?  What if you need to change/update the hover text?

revarbat, 2023-04-09:
I do already have hover-text `STD` FileFootnotes: tags in the wiki Sidebar.  They are not currently implemented in the cleanest way, however.  I am thinking of implementing a DefineFileFootnotes: configuration block that will let you do something like this in the config file, though:

```
DefineFileFootnotes:
    STD=Included in std.scad
    Parts=Can create parts.
    MAT=Support for transformation matrices.
```

Then in each file:

```
FileFootnotes: STD,MAT
```

Similarly, I am thinking of `DefineTags:` and `Tags:` for Constants/Function/Module documentation. The tags can be shown in the same places in the indices that we currently show Synopsis: line text.

I should note that we currently do have a number of Function&Modules that have a Topics entry for "VNF Generator".

RAMilewski, 2023-04-09:
For repetitive hover texts like [VNF] you can just input a placeholder such as xyzzy and then do a search and replace to put in the standard hover text syntax everywhere it's used in a file.   If there's a way for docsgen to do the replacing then that's a win.  

Personally, I favor a hover text rather than a footnote link that takes the reader away from the module or function in question.   

FWIW the STD links in the sidebar  don't appear to point to actual content yet.   ...at least not in Firefox or Chrome.

adrianVmariano, 2023-04-09:
If I write a new function next month and need to put 3 hover texts in its summary then I cannot use your search and replace scheme.  That's what I mean about short term thinking.   Yeah, if the library is done, never more to change, and we're writing the final documentation, then your approach of search and replace works.  But if it's a dynamic work, that's going to keep changing, and you need to be able to write and rewrite the tags in the future, then it does not.

RAMilewski, 2023-04-09:
I'm all for letting docsgen do the fiddly bits!  So let's define what the mouse-over links look like before we write any more synopses.

[vnf], [geom], and [vnf || geom]  perhaps?

adrianVmariano, 2023-04-09:
The compound idea [vnf || geom] is a mistake.  It seems like we need at least [vnf], [geom], [path], [rgn] (or [region]).   Not sure if that covers it?  I don't know how far we want to push this.  Like we could have things like [list] or [array] or [number], [boolean]....

RAMilewski, 2023-04-09:
Not sure the compound idea is worthless. Since we can define each different mouseover only once at the beginning of the file, having some common compound mouseover links would simplify the synopses.
It's a little hard to show here because we're inside Github Markdown, but....

This code (minus the backslashes) near the top of the markdown:
`
  \[vnf]: ## "Used as a function it returns a vnf."

  \[geom]: ## "Used as a function it returns geometry."

  \[vnf||geom]: ## "Used as a function it can return a vnf or geometry."

  \[mat]: ## "Used as a function it returns a transformation matrix."
`

Works like this: 

  [vnf]: ## "Used as a function it returns a vnf."
  [geom]: ## "Used as a function it returns geometry."
  [vnf||geom]: ## "Used as a function it can return a vnf or geometry."
  [mat]: ## "Used as a function it returns a transformation matrix."

Synopsis: This module creates an elephant. [VNF][vnf] 

Synopsis: This module creates a tiger. [more info][vnf||geom]

RAMilewski, 2023-04-09:
Hmmm... github got me again!   The Synopsis lines as written look like:

Synopsis: This module creates an elephant. [VNF][vnf] 

Synopsis: This module creates a tiger. [more info][vnf||geom]

RAMilewski, 2023-04-09:
Complication:  None of this survives docsgen...   yet.

adrianVmariano, 2023-04-09:
A function cannot return geometry.  So the mouseover for [geom] should be "Can be used as a module to return geometry" and then the mouseover for [vnf] would be "Can be used as a function to return a VNF".  The problem with compound entries is that then you have to worry about all possible combinations.  What if something can be geometry, or return a VNF or transformation matrix?  What if it returns a path or vnf or geometry?   I think each return should be separate.  

There is a question of whether there should be [geom2d] "can be used as a module to return 2d geometry" and [geom3d] analogously.  Or maybe just [2d] and [3d] would work for that?

RAMilewski, 2023-04-09:
The synopsis describes the most common usage, with the mouseovers providing completeness.
For many entries the synopsis is describing the module. 
You know the library far better than I, which module or function has the largest number of different returns?

adrianVmariano, 2023-04-09:
At the moment I can't think of anything with more than 3 return cases.  (path_sweep can return geometry as a module or it can return a vnf or a transform list as a function.)    I think you maybe can get a list of paths out of vnf_halfspace in addition to a vnf.  Not sure actually if there is a module.  (Maybe if there isn't there should be?)

I see no advantage of having the combined mouseover, though.

RAMilewski, 2023-04-09:
> I see no advantage of having the combined mouseover, though.

Fewer things for the user to click on.

[32 more replies]

closed 2023-05-04 by adrianVmariano