BelfrySCAD/BOSL2 #1085: [BUG]

dan-p3rry, 2023-03-21:
screws.scad is hanging with M12 screws.  I can only shut down OpenSCAD to re-try my code.

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>
include <BOSL2/screws.scad>

$fn=90;
$slop = 0.05;
sb_screw();

module sb_screw() {
    
    spec = screw_info("M12x1, 8", head = "socket");
    //spec = screw_info("M6x1,12",head="socket");
    newspec = struct_set(spec,["head_size", 20, "head_height", 3]);
    screw(newspec);
}

**Expected behavior**
It works correctly with the example from documentation, for M6 screw

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 2023.02.15
- Other libraries used:

dan-p3rry, 2023-03-21:
I think I found the issue, it breaks when there is a space in the string "M12x1, 8".  "M12x1,8" works.  It's fine to require a particular syntax for a function to work, but it shouldn't crash OpenSCAD....  Thanks.

adrianVmariano, 2023-03-21:
I tried this in the stable version 2021.01 and in the dev I had installed, 2022.12.07.  In both cases, I get an error message, not a crash.  

```
[WARNING: undefined operation (undefined >= number) in file lib/BOSL2/strings.scad, line 404](https://github.com/revarbat/BOSL2/issues/404,/home/adrian/scad/lib/BOSL2/strings.scad)
[TRACE: called by '_parse_int_recurse' in file lib/BOSL2/strings.scad, line 478](https://github.com/revarbat/BOSL2/issues/478,/home/adrian/scad/lib/BOSL2/strings.scad)
[TRACE: called by 'parse_frac' in file lib/BOSL2/strings.scad, line 502](https://github.com/revarbat/BOSL2/issues/502,/home/adrian/scad/lib/BOSL2/strings.scad)
[TRACE: called by 'parse_num' in file lib/BOSL2/screws.scad, line 1310](https://github.com/revarbat/BOSL2/issues/1310,/home/adrian/scad/lib/BOSL2/screws.scad)
[TRACE: called by '_parse_screw_name' in file lib/BOSL2/screws.scad, line 1788](https://github.com/revarbat/BOSL2/issues/1788,/home/adrian/scad/lib/BOSL2/screws.scad)
[TRACE: called by 'screw_info' in file screwfail.scad, line 10](https://github.com/revarbat/BOSL2/issues/10,/home/adrian/scad/screwfail.scad)
[TRACE: assignment to 'spec' in file screwfail.scad, line 10](https://github.com/revarbat/BOSL2/issues/10,/home/adrian/scad/screwfail.scad)
[TRACE: called by 'sb_screw' in file screwfail.scad, line 6](https://github.com/revarbat/BOSL2/issues/6,/home/adrian/scad/screwfail.scad)
Execution aborted
```

It's a bad error message, but better than a crash.   The problem is due to parse_frac() not handling a leading space correctly.  So I've fixed that.  Now I get:

```
[ERROR: Assertion '((len(commasplit) == 1) || is_num(length))' failed: "Invalid length " 8" in screw name, "M12x1, 8"" in file lib/BOSL2-fork/screws.scad, line 1331](https://github.com/revarbat/BOSL2/issues/1331,/home/adrian/scad/lib/BOSL2-fork/screws.scad) 
[TRACE: called by '_parse_screw_name' in file lib/BOSL2-fork/screws.scad, line 1802](https://github.com/revarbat/BOSL2/issues/1802,/home/adrian/scad/lib/BOSL2-fork/screws.scad) 
[TRACE: called by 'screw_info' in file screwfail.scad, line 11](https://github.com/revarbat/BOSL2/issues/11,/home/adrian/scad/screwfail.scad) 
[TRACE: called by 'sb_screw' in file screwfail.scad, line 6](https://github.com/revarbat/BOSL2/issues/6,/home/adrian/scad/screwfail.scad)
```
But still need to figure out why OP got a crash.  OpenSCAD bug in the latest dev?

adrianVmariano, 2023-03-21:
I figured out the cause of the crash:  OP does not have "stop on first warning" turned on.  With this off, there is an infinite recursion.   The stable version catches this:

```
[WARNING: undefined operation (undefined >= number) in file lib/BOSL2-fork/strings.scad, line 404](https://github.com/revarbat/BOSL2/issues/404,/home/adrian/scad/lib/BOSL2-fork/strings.scad) 
[WARNING: undefined operation (undefined >= number) in file lib/BOSL2-fork/strings.scad, line 404](https://github.com/revarbat/BOSL2/issues/404,/home/adrian/scad/lib/BOSL2-fork/strings.scad) 
[WARNING: undefined operation (undefined >= number) in file lib/BOSL2-fork/strings.scad, line 404](https://github.com/revarbat/BOSL2/issues/404,/home/adrian/scad/lib/BOSL2-fork/strings.scad) 
[WARNING: undefined operation (undefined >= number) in file lib/BOSL2-fork/strings.scad, line 404](https://github.com/revarbat/BOSL2/issues/404,/home/adrian/scad/lib/BOSL2-fork/strings.scad) 
[WARNING: undefined operation (undefined >= number) in file lib/BOSL2-fork/strings.scad, line 404](https://github.com/revarbat/BOSL2/issues/404,/home/adrian/scad/lib/BOSL2-fork/strings.scad) 
[ERROR: Recursion detected calling function 'search' in file lib/BOSL2-fork/strings.scad, line 403](https://github.com/revarbat/BOSL2/issues/403,/home/adrian/scad/lib/BOSL2-fork/strings.scad) 
[TRACE: called by '_parse_int_recurse' in file lib/BOSL2-fork/strings.scad, line 406](https://github.com/revarbat/BOSL2/issues/406,/home/adrian/scad/lib/BOSL2-fork/strings.scad) 
[TRACE: called by '_parse_int_recurse' in file lib/BOSL2-fork/strings.scad, line 406](https://github.com/revarbat/BOSL2/issues/406,/home/adrian/scad/lib/BOSL2-fork/strings.scad) 
[TRACE: called by '_parse_int_recurse' in file lib/BOSL2-fork/strings.scad, line 406](https://github.com/revarbat/BOSL2/issues/406,/home/adrian/scad/lib/BOSL2-fork/strings.scad) 
[TRACE: called by '_parse_int_recurse' in file lib/BOSL2-fork/strings.scad, line 406](https://github.com/revarbat/BOSL2/issues/406,/home/adrian/scad/lib/BOSL2-fork/strings.scad) 
[TRACE: called by '_parse_int_recurse' in file lib/BOSL2-fork/strings.scad, line 406](https://github.com/revarbat/BOSL2/issues/406,/home/adrian/scad/lib/BOSL2-fork/strings.scad)
```

but the dev version does not.  So the crash is an OpenSCAD bug in the dev version.

adrianVmariano, 2023-03-23:
The BOSL2 bug is fixed in the PR.  I don't know if we need to try to make a minimal case to report as an OpenSCAD bug.

dan-p3rry, 2023-03-23:
Thanks, Adrian.  That was very impressive debugging.
Regards,
Dan

adrianVmariano, 2023-03-25:
I made a reduced case and filed a bug against OpenSCAD for this.  Closing the issue now.

closed 2023-03-25 by adrianVmariano