BelfrySCAD/BOSL2 #1092: trapezoid anchoring inconsistency

adrianVmariano, 2023-03-25:
Trapezoid CENTER anchoring is different for the function than for the module.  

```polygon(trapezoid( w1=100, h=40,w2=70,shift=33));```

[image]

```trapezoid( w1=100, h=40,w2=70,shift=33);```

[image]

The CENTER anchor is at the midpoint between the FRONT and BACK anchors, but the trapezoid is not created in its centered position.  It appears that reorient assumes that the native position is the CENTER position.  But also, that assumption seems to be important because various modules call the function form, so they might get confused if that isn't true.   If I take out the line that special cases the CENTER anchor in reorient() then the anchors are still different from each other, just with the trapezoid in a different place.  

Do we need a special anchor like "origin" to use to handle this general situation?  (That is, modules that call the function request "origin" anchoring which does what CENTER does now, and CENTER is changed to use the standard anchor procedure).

adrianVmariano, 2023-03-28:
@revarbat Did you fix this when you fixed prismoid anchoring?

revarbat, 2023-03-29:
I have not touched the 2D `trapezoid()`, so I believe this is not yet fixed.

adrianVmariano, 2023-03-29:
Hmmm.   If you change reorient and didn't fix trapezoid() it will be producing displaced anchors.

adrianVmariano, 2023-03-29:
Is this fixed now?

revarbat, 2023-03-31:
This is fixed.

closed 2023-03-31 by revarbat