BelfrySCAD/BOSL2 #1106: Redundant Masks?

RAMilewski, 2023-03-30:
The ang parameter in rounding_angled_edge_mask() defaults to 90°.  Doesn't that make rounding_edge_mask() redundant?

Should rounding_edge_mask() be dropped and rounding_angled_edge_mask() be renamed?

Same for rounding_corner_mask().

revarbat, 2023-04-03:
This is a reasonable idea.

adrianVmariano, 2023-10-11:
The actual shapes produced are different.  Does it matter?  For some reason rounding_angled_edge_mask makes a shape that rounds on two sides.  

This mask shape might also be used to create a fillet (that is, added to the model instead of removed) in which case the double-sided one would be bad.

RAMilewski, 2023-10-12:
For the rounding corner masks, the orientation is different as well.  

In example 2, rounding_corner_mask() is used with corner_mask().  To do the same with rounding_angled_corner_mask() you need to add an xrot(180) between the two.

adrianVmariano, 2023-11-21:
So does this mean we should leave things alone instead of merging the masks?

closed 2024-05-12 by revarbat