BelfrySCAD/BOSL2 #1116: `list_set` does not honor negative indexes

TwoClocks, 2023-04-01:
I would expect this:
```
echo(list_set([1,2,3,4],-1,7));
```
to yield ` [1, 2, 3, 7]` but it seems to do nothing and yields ` [1, 2, 3, 4]`

adrianVmariano, 2023-04-02:
Right now only `slice` and `select` accept negative indices.   So various questions arise:

1. Should this change?  Are there any concerns here or ways that this would create bugs or break anything?  
2. What is the complete list of functions that would need to be updated?   
3. What is current behavior in these various functions with negative indices?  (Is it useful?)
4. What would the correct behavior be for `list_set([1,2,3,4], -33, 7)`, for example?  I lean towards thinking it's an error.  Is that right?

TwoClocks, 2023-04-02:
Since I wrote this, I've moved to solidPython, so all my list management is being done there.

For the time being, I'd at least make `set_list` error w/ a negative index. Same for everything except `slice` and `select`.

I'd have the behavior be the same for anything that will accept a negative index.

revarbat, 2023-04-03:
I am ambivalent about whether to support negative indices here.  If we do not, however, at the least we should assert a non-negative index, so long as it does not affect efficiency in a bad way. I don't think `list_set()` is a frequently used call.

revarbat, 2023-04-03:
Actually, thinking about it, this may be an indication that we need a standard call that asserts that a given index is not out of range for a given list.  Something like:

revarbat, 2023-04-03:
Secondarily, this may also indicate that we could use a user accessible function to convert a wrappable/negative index for a list into an actual in-bounds index.

adrianVmariano, 2023-04-03:
I can't imagine a user for a function that remaps indices when you can just use select instead.   Or you can do `L%len(list)` which is not going to be any longer than some call.  In fact, shorter.   What would the alternative be?  `wrap_idx(L, list)`?

Possibly the list assertion is problematic because it adds an extra function call to a possibly core function.  Not sure.  Note that in the case raised in this issue of list_set, values beyond the end of the list are valid.  

I did look at the uses of list_set in the library and there is one that would benefit from negative indexing.

closed 2023-05-18 by revarbat