BelfrySCAD/BOSL2 #1131: VNF from text3d

elxris, 2023-04-06:
**Is your feature request related to a problem? Please describe.**
Documentation states that from text() or text3d() we can get an vnf but no function is defined.

**Describe the solution you'd like**
If possible, make that happen.

**Describe alternatives you've considered**
I don't think there are alternatives to that. As some functions are very useful with vnf structures.
I've considered to process the font externally to get the 2D paths.

**Example Code**
```
vnf = text3d("example text");
```

**Additional context**
No additional context

adrianVmariano, 2023-04-06:
This is not possible, at least as things stand.  There is no way to get the information out of geometry, so when openscad creates text, that's it---there is no way to access it.  So you would have to do like you said and produce the text externally somehow.

adrianVmariano, 2023-04-06:
The synopsis needs to corrected not to say that you can get a VNF.

elxris, 2023-04-06:
@adrianVmariano yes, it should be updated as text(), text3d(), and path_text() all says that you can access the vnf. Thank you for answering. I hope this is possible in the future.

adrianVmariano, 2023-04-06:
Thanks for identifying the offenders...though I can't find such a reference for text().   I fixed the other two and pushed a PR.

elxris, 2023-04-07:
You're right, I think I did hallucinate that one.

closed 2023-04-08 by adrianVmariano