BelfrySCAD/BOSL2 #1133: [BUG] Slider does not match rail when you change the rail width.

Rickrat, 2023-04-09:
**Describe the bug**
Cannot change the width of the rail, otherwise the slider does not match the rail.

**Code To Reproduce Bug**
```
translate([0, 0, 0]) rail(l=150, w=20, h=10, chamfer=2, ang=15);
translate([40, 0, 0]) slider(l=30, base=10, wall=4, $slop=0.2, spin=90);

```

**Expected behavior**
It should match.

revarbat, 2023-05-18:
Err, you forgot to specify `w=20` in the second line.

revarbat, 2023-05-18:
There may still be a dimensional mismatch, though.

revarbat, 2023-05-18:
Nevermind.  Also missing `ang=15` in the second line.

```openscad
rail(l=150, w=20, h=10, chamfer=2, ang=15);
up(20) zflip() slider(l=30, w=20, ang=15, base=10, wall=4, $slop=0.2, spin=90);
```
[image]

closed 2023-05-18 by revarbat