BelfrySCAD/BOSL2 #1152: metric_screws transition

adrianVmariano, 2023-05-01:
We need to figure out what's happening with metric_screws.scad.  Do we make it a deprecated/undocumented passthrough to screws.scad for compatibility?  Is that easy to do?

revarbat, 2023-05-01:
I think that screws.scad can do everything that metric_screws.scad can.  I want to remove metric_screws.scad as being redundant.  The easiest solution would be to remove metric_screws.scad from documentation, but leave the file intact for importing, with the existing calls deprecated while mentioning the new screws.scad.  Which is more or less what I think you propose above.

adrianVmariano, 2023-05-01:
I guess there are a few options with varying degrees of complication:

1. Leave metric_screws intact but hide it and do nothing else
2. Leave metric_screws intact but have a vague deprecation message appear if any of its code is called (e.g. use a $ variable to track if anything is called and on first call, print some message like "metric_screws.scad has been replaced by screws.scad and will be removed in a future version"
3. Leave metric_screws intact but give more helpful advice, like "module foo in metric screws has been replaced by baz in screws.scad and call it like ...."
4. Recode everything in metric_screws to be an actual pass-through to screws.scad and print helpful messages

I guess I was vaguely thinking about doing 4, but doing 2 would be easier.  

I was looking at what's there:  if someone is relying on the various functions then there's not a simple drop-in replacement---you have to use screw_info and then extract the appropriate field, which is arguably a bit of a pain.

revarbat, 2023-06-02:
I'm thinking hide metric_screws.scad from the documentation, but leave the file intact, and have a file-level ECHO that notes the deprecation of metric_screws.scad, so that it just shows the message at include time.  I don't think there's a way to set an $var to show a message only once.

PR #1185 (2023-06-02): Deprecating metric_screws.scad
Resolves #1152
closed 2023-06-02 by revarbat