BelfrySCAD/BOSL2 #1184: Additional texture - no texture

SnowballsChance, 2023-06-01:
I would like to be able to cleanly turn off the texture on a cylinder during the preview because it is slow on my computer, once I have decided on the texture settings and am happy with it, I really don't need to see the texture while I am debugging and adjusting other things.

I'd like to be able to do something like this:

```
cyl_texture = texture("trunc_ribs_vnf", gap = 0.5, inset = 1 / 10);
no_texture = texture("none");

cyl(d = outer_diameter, height = main_cylinder_height - floor_thickness * 2, chamfer = cyl_chamfer, texture = $preview ? no_texture : cyl_texture, tex_scale = 0.5, tex_size = [ 5, 5 ], $fn = $preview ? lowres : hires, anchor = BOTTOM);
```

I usually use an if( preview is true) 

```
if ($preview)
   {
      cyl(d = outer_diameter, height = main_cylinder_height - floor_thickness * 2, chamfer = cyl_chamfer,
      tex_scale = 0.5, tex_size = [ 5, 5 ], $fn = lowres , anchor = BOTTOM);
   } 
else
   {
     cyl(d = outer_diameter, height = main_cylinder_height - floor_thickness * 2, chamfer = cyl_chamfer,
      texture = cyl_texture, tex_scale = 0.5, tex_size = [ 5, 5 ], $fn = hires, anchor = BOTTOM);
    }
```

The small problem with this is now I need to ensure both paths of code are the same but for the texture and $fn.

Thanks for considering this.

If I work out how to add it to skin.scad I will post it here.

revarbat, 2023-06-02:
Note that `texture=undef` means "Don't texture."

Try a design pattern more like:

    cyl_texture = $preview? undef : texture("trunc_ribs_vnf", gap = 0.5, inset = 1 / 10);
    cyl(d = outer_diameter, height = main_cylinder_height - floor_thickness * 2, chamfer = cyl_chamfer, texture = cyl_texture, tex_scale = 0.5, tex_size = [ 5, 5 ], $fn = $preview ? lowres : hires, anchor = BOTTOM);

revarbat, 2023-06-02:
Mind you that not texturing will not construct the cylinder in a slightly different way than texturing will.  You might get a slightly more representative preview using `[[0]]` instead of `undef`.

SnowballsChance, 2023-06-02:
Thank you - I appreciate your help! - I will have a go with both of those options. 

So very many thanks!

closed 2023-06-02 by revarbat

SnowballsChance, 2023-06-02:
I gave it a go and yep works exactly as suggested. Thanks again!

This whole issue was triggered because I have been experiencing very slow renders and pretty poor performance once a model reaches any moderate level of complexity. 
I have an M1 Mac and noticed that the release of OpenSCAD I have been using runs as an Intel binary thus an added level of translation related lag. Further down the OpenSCAD [downloads](https://openscad.org/downloads.html) page are the development snapshots which run natively on the M1 and so very much faster.  So doing a full render went from 22 minutes to 4 minutes! and the low quality renders are equally quick.