BelfrySCAD/BOSL2 #1188: Add support for planetary gears

jon-bondy, 2023-06-17:
It would be helpful if one could design a set of planetary gears using BOSL.

adrianVmariano, 2023-06-17:
So specifically you mean that you want ring gears, or do you mean something else?  That is, what exactly would this support do?

revarbat, 2023-06-17:
As I understand it, only the ring gear is maybe confusing to make, and that can be done by making a `spur_gear()` with `interior=true` then differencing it away from a cylinder.

adrianVmariano, 2023-06-17:
There should be an example showing that this is possible.   Is there any way to discover this in the manual?

revarbat, 2023-06-17:
```openscad
include <BOSL2/std.scad>
include <BOSL2/gears.scad>
$fa=1; $fs=1;
diff()
cyl(d=100,l=10)
    tag("remove")
        spur_gear(
            pitch=8, teeth=32, thickness=11,
            pressure_angle=20, interior=true);

color("green")
back(27)
spur_gear(
    pitch=8, teeth=11, thickness=11,
    pressure_angle=20);
```
[image]

adrianVmariano, 2023-06-17:
Shouldn't the teeth on the ring gear look the same as the teeth on the spur gear?

revarbat, 2023-06-17:
Let me see if I can whip up a proper planetary gear assembly with animation.

adrianVmariano, 2023-06-17:
That might be overkill.  The question isn't whether the above thing will work ok in an animation, but whether it's actually the right way to make a ring gear.

revarbat, 2023-06-18:
No, the interior teeth of a ring gear do NOT look like inward pointing spur gear teeth.  The animation I just make makes that fairly clear.

https://github.com/BelfrySCAD/BOSL2/assets/350001/106b6ae8-d36f-40d3-ae55-d71610092959

revarbat, 2023-06-18:
Oh, the code for the above animation:
```openscad
include <BOSL2/std.scad>
include <BOSL2/gears.scad>
$fa=1; $fs=1;
diff()
cyl(d=100,l=10)
    tag("remove")
        spur_gear(
            pitch=8, teeth=32, thickness=11,
            pressure_angle=20, interior=true);

color("green")
back(27)
spur_gear(
    pitch=8, teeth=11, thickness=11,
    pressure_angle=20);
```

adrianVmariano, 2023-06-18:
There is an outstanding issue about how we need actual docs for the gears stuff.  It would definitely help.  I mean, we need a multi-page gears intro section that explains all the terminology and gear types and gear parts with pictures.  But for now I made a few tweaks to the docs that should help a bit.  I added an example of non-parallal-shaft helical gears----but I couldn't figure out how to reliably get them to mesh in general.  Do you know the answer?  

I added a reference to internal ring gears to the spur_gear module.  It appears that technically helical gears are not a subtype of spur gears, so the name isn't ideal, though I guess it's a tradition in OpenSCAD to do that.  

I wonder about how to make herringbone or double helical gears.  The one gear example I printed using herringbone gears.  It should be an easy development from what we have, but just stacking to gears, they weren't aligned properly at the joint.   I'm not sure if a separate module makes sense?   Or a `herringbone=true` or `double=true` option?   I guess doing these really well would mean rounding the teeth where they meet in the center of the gear, which would be much harder than just sticking together two gears from the existing code.  

In the case of bevel gears, the same question arises, actually.  The other thing with bevel gears is that there is an interior option.  But I couldn't figure out what an internal bevel gear is.   The interior option really should be internal, but maybe too late to change?  I don't know.   And the question also arises of whether internal should have a default_tag of "remove".  

Also it appears that you can't make the worm curve to match the worm gear for the "double enveloping" assembly.  

So here are some questions:

1. how do you determine meshing for non-parallel shaft helical gears
2. how do you make herringbone aka double helical gears?  Or the analgous spiral bevel gear construction? 
3. What is an interior bevel gear?  
4. Should "interior" change to "internal"?
5. Should interior gears have a default tag of "remove"
6. Double enveloping worm gear assembly?
7. Any idea about zerol bevel gears?

adrianVmariano, 2023-06-18:
```
spur_gear(
       pitch=pitch, teeth=n, thickness=10,
       shaft_diam=5, helical=0, slices=12,
       $fa=1, $fs=1
   ) 
  position(RIGHT)
   spur_gear(
       pitch=pitch, teeth=n, thickness=10,
       shaft_diam=5, helical=0, slices=12,
       $fa=1, $fs=1,anchor=LEFT
   ) 
;
```

Above code makes two mated gears (I think) but it's impossible to rotate the gears with the above code.  There are two problems.  One is that rotating an object rotates in children and the other is that spin= rotates relative to the anchor instead of object center (a design decision that has always seemed wacky to me).   Is there a way to assemble gears using positioning, or is that just an idea that doesn't work?   Do gears need to have an option to offset their teeth so they can be aligned?

jon-bondy, 2023-06-18:
I've been playing around with the code, and from a user's perspective, 
what is wanted is to specify the OD and the mechanical advantage and 
perhaps the diameter of the central gear.  I guess the question is how 
much effort do you want to put into making this simple for a library user.

Jon

--------------sQLPdxueG0cOljOBhkQQDILs
Content-Type: text/html; charset=UTF-8
Content-Transfer-Encoding: 8bit

<html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
  </head>
  <body>
    <p>I've been playing around with the code, and from a user's
      perspective, what is wanted is to specify the OD and the
      mechanical advantage and perhaps the diameter of the central
      gear.  I guess the question is how much effort do you want to put
      into making this simple for a library user.</p>
    <p>Jon<br>
    </p>
    <div class="moz-cite-prefix">On 6/17/2023 11:52 PM, Revar Desmera
      wrote:<br>
    </div>
    <blockquote type="cite" ***@***.***">
      
      <p dir="auto">Oh, the code for the above animation:</p>
      <div class="highlight highlight-source-scad" dir="auto">
        <pre class="notranslate"><span class="pl-k">include</span> <span class="pl-k">&lt;</span>BOSL2/std.scad<span class="pl-k">&gt;</span>
<span class="pl-k">include</span> <span class="pl-k">&lt;</span>BOSL2/gears.scad<span class="pl-k">&gt;</span>
<span class="pl-k">$</span>fa<span class="pl-k">=</span><span class="pl-c1">1</span>; <span class="pl-k">$</span>fs<span class="pl-k">=</span><span class="pl-c1">1</span>;
diff()
cyl(d<span class="pl-k">=</span><span class="pl-c1">100</span>,l<span class="pl-k">=</span><span class="pl-c1">10</span>)
    tag(<span class="pl-s">&quot;remove&quot;</span>)
        spur_gear(
            pitch<span class="pl-k">=</span><span class="pl-c1">8</span>, teeth<span class="pl-k">=</span><span class="pl-c1">32</span>, thickness<span class="pl-k">=</span><span class="pl-c1">11</span>,
            pressure_angle<span class="pl-k">=</span><span class="pl-c1">20</span>, interior<span class="pl-k">=</span><span class="pl-c1">true</span>);

<span class="pl-c1">color</span>(<span class="pl-s">&quot;green&quot;</span>)
back(<span class="pl-c1">27</span>)
spur_gear(
    pitch<span class="pl-k">=</span><span class="pl-c1">8</span>, teeth<span class="pl-k">=</span><span class="pl-c1">11</span>, thickness<span class="pl-k">=</span><span class="pl-c1">11</span>,
    pressure_angle<span class="pl-k">=</span><span class="pl-c1">20</span>);</pre>
      </div>
      <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
          visibility: hidden; overflow: hidden; opacity: 0; width: 0;
          height: 0; max-width: 0; max-height: 0; mso-hide: all">Message
          ID: <span>&lt;BelfrySCAD/BOSL2/issues/1188/1595942221</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
      <script type="application/ld+json">[
{
"target": "https://github.com/BelfrySCAD/BOSL2/issues/1188#issuecomment-1595942221",
"url": "https://github.com/BelfrySCAD/BOSL2/issues/1188#issuecomment-1595942221",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
    </blockquote>
  </body>
</html>

--------------sQLPdxueG0cOljOBhkQQDILs--

adrianVmariano, 2023-06-18:
Not sure what you mean about specifying "the diameter of the central gear".   Are you thinking of a whole assembly of gears?  We don't have any code in the library for making gear assemblies, just individual gears.

jon-bondy, 2023-06-18:
Yes.  Sorry.  The diameter of the assembly.  It will quite likely have 
to fit into something else.

Worm gears and worms seem like an assembly.

The planetary gear set seems like an assembly.

Put differently, if I were to design the central, planetary, and 
outer/ring gears separately, how would I know they would make a 
successful assembly absent a single central module() which created them 
as a set?

Your library, your rules.  Just my thoughts.

Jon

--------------7lD0AvUgE3gip9uiiXMckvJ0
Content-Type: text/html; charset=UTF-8
Content-Transfer-Encoding: 8bit

<html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
  </head>
  <body>
    <p>Yes.  Sorry.  The diameter of the assembly.  It will quite likely
      have to fit into something else.</p>
    <p>Worm gears and worms seem like an assembly.</p>
    <p>The planetary gear set seems like an assembly.</p>
    <p>Put differently, if I were to design the central, planetary, and
      outer/ring gears separately, how would I know they would make a
      successful assembly absent a single central module() which created
      them as a set?</p>
    <p>Your library, your rules.  Just my thoughts.</p>
    <p>Jon<br>
    </p>
    <div class="moz-cite-prefix">On 6/18/2023 10:55 AM, adrianVmariano
      wrote:<br>
    </div>
    <blockquote type="cite" ***@***.***">
      
      <p dir="auto">Not sure what you mean about specifying &quot;the
        diameter of the central gear&quot;. Are you thinking of a whole
        assembly of gears? We don't have any code in the library for
        making gear assemblies, just individual gears.</p>
      <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
          visibility: hidden; overflow: hidden; opacity: 0; width: 0;
          height: 0; max-width: 0; max-height: 0; mso-hide: all">Message
          ID: <span>&lt;BelfrySCAD/BOSL2/issues/1188/1596175643</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
      <script type="application/ld+json">[
{
"target": "https://github.com/BelfrySCAD/BOSL2/issues/1188#issuecomment-1596175643",
"url": "https://github.com/BelfrySCAD/BOSL2/issues/1188#issuecomment-1596175643",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
    </blockquote>
  </body>
</html>

--------------7lD0AvUgE3gip9uiiXMckvJ0--

adrianVmariano, 2023-06-18:
You should try to figure out how to stop putting HTML into your responses, because it makes the github issue look terrible.  

I have no objection to the idea of supporting gear assemblies, at least in principle, but it's not something I know anything about.  The question that then arises is what is the catalog of all possible gear assemblies that should be supported?   How does this not turn into a catalog of 1036 different gear assemblies?  

I guess another question might be whether there's some way to make it easier to design working assembles with the right characteristic from the individual gears.  I mean, assuming that it's not already easy.  I don't know.   In case this isn't clear, Revar wrote all the gear code I basically don't know anything about it.

revarbat, 2023-06-19:
Herringbone gear:

```openscad
zflip_copy()
    spur_gear(
        pitch=5, teeth=20, thickness=10,
        shaft_diam=5, helical=+30, slices=10,
        anchor=BOT);
```

[image]

revarbat, 2023-06-19:
I honestly am unclear why stacking helical gears with opposite helical angles don't align.  The code seems like it should work, but it obviously doesn't.

revarbat, 2023-06-19:
Okay, so helical gears use twist= in `linear_extrude()` to implement their helical shape.  Which means that the twist at the bottom of the gear is always 0, and the top is at the full twist.  So if you just stack gears, you only get a match if you are lucky.  I interpret this as meaning that helical spur gears need to be rotated back by half the twist so that the middle of the gear is at zero-twist.

adrianVmariano, 2023-06-19:
Not sure why flipping didn't occur to me.   There should be an example....

revarbat, 2023-06-19:
I'll add a herringbone gear example with the twist fix.

adrianVmariano, 2023-06-19:
You should merge my stuff before changing more to avoid conflict issues

adrianVmariano, 2023-06-19:
I think I fixed the conflict.

adrianVmariano, 2023-06-19:
Would it be possible to make the herringbone with a short section of straight spur gear in the middle to avoid the sharp corner?

adrianVmariano, 2023-06-19:
Remaining questions:

1.  how do you determine meshing for non-parallel shaft helical gears
2.  What is an interior bevel gear?
3.  Should "interior" change to "internal"?
4.  Should interior gears have a default tag of "remove"
5.  Double enveloping worm gear assembly?
6.  Any idea about zerol bevel gears?

Actually also, is there some reason to make the herringbone gear the way it's done in the example, with two spur_gear() calls?  It seems like using zflip_copy() would be simpler, and could create a whole assembly of gears all at once.

revarbat, 2023-06-20:
1. This is not a simple topic, and needs much more than a few lines of explanation.
2. I don't think an interior bevel gear is a thing that exists.  `interior=` is not a valid argument for `bevel_gear()`
3. I have maybe a slight preference for `internal=` over `interior=`.
4. Maybe an interior gear should have `default_tag("remove")`.  Or maybe we need to create a `ring_gear()` module.  I'm undecided.
5. I have no idea what a double enveloping worm gear assembly is.
6. I think in theory, a zerol bevel gear just has `spiral_angle=0`

adrianVmariano, 2023-06-29:
@jon-bondy What would a planetary gear module do, exactly.  I fiddle around with planetary gearing and wrote the following code that attempts to produce a gear with a specific sun/ring ratio.  This is a bit tricky, since it seems that the ratio needs to feature 2 odd numbers in order to produce a valid gear assembly.  And you probably don't want 6000 teeth.  

```
include<BOSL2/std.scad>
include<BOSL2/gears.scad>

$fa=1;
$fs=1;

pitch=3;

N=4;             // number of planetary gears
maxteeth=55;     // max teeth on ring gear
ratio = 1/5.123; // desired ratio between ring and sun gear (fixed carrier)

///////////////////////////////////////////////////////

pq = rational_approx(ratio, maxteeth);
factor = floor(maxteeth/pq[1]);
temp_z1 = factor*pq[0];
temp_z3 = factor*pq[1];
z1 = temp_z1%2==0 ? temp_z1+1 : temp_z1;
z3 = temp_z3%2==0 ? min(temp_z3+1, maxteeth-(maxteeth%2==0?1:0)) : temp_z3;
z2 = (z3-z1)/2;
assert(z2==floor(z2));

echo(z1=z1,z3=z3, req_ratio=ratio, actual_ratio=z1/z3, gear_gcd=gcd(z1,z3));

r1 = pitch_radius(pitch,z1);
r2 = pitch_radius(pitch,z2);
r3 = pitch_radius(pitch,z3);

//rot(-90+ang)
  spur_gear2d(pitch=pitch, teeth=z1);

quant = 360/(z1+z3);

diff()
zrot(360/z3/2 * (1-(z1%2)))
circle(r=r3+3)
tag("remove")spur_gear2d(pitch=pitch, teeth=z3, internal=true);

for (uang=lerpn(0,360,N,endpoint=false))
  let(ang = quant(uang,quant),f=echo(ang))
  move( (r1+r2)*[cos(ang),sin(ang)])rot((z1/z2)*(ang-90))rot(90+ang+360/z2/2)spur_gear2d(pitch=pitch, teeth=z2);

// Function rational_approx()
// Usage:
//   pq = rational_approx(x, maxq);
// Description:
//   Finds the best rational approximation p/q to the number x so that q<=maxq.  Returns
//   the result as `[p,q]`.  If the input is zero, then returns `[0,1]`.  
// Example:
//   pq1 = rational_approx(PI,10);    // Returns: [22,7]
//   pq2 = rational_approx(PI,10000); // Returns: [355, 113]
//   pq3 = rational_approx(221/323);  // Returns: [13,19]
//   pq4 = rational_approx(0);        // Returns: [0,1]
function rational_approx(x, maxq, cfrac=[], p, q) =
  let(
       next = floor(x),
       fracpart = x-next,
       cfrac = [each cfrac, next],
       pq = _cfrac_to_pq(cfrac)
  )
    approx(fracpart,0) ? pq 
  : pq[1]>maxq ? [p,q]
  : rational_approx(1/fracpart,maxq,cfrac, pq[0], pq[1]);

// Converts a continued fraction given as a list with leading integer term
// into a fraction in the form p / q, returning [p,q]. 
function _cfrac_to_pq(cfrac,p=0,q=1,ind) =
    is_undef(ind) ? _cfrac_to_pq(cfrac,p,q,len(cfrac)-1)
  : ind==0 ? [p+q*cfrac[0], q]
  : _cfrac_to_pq(cfrac, q, cfrac[ind]*q+p, ind-1);
```

jon-bondy, 2023-06-29:
I tried to disable HTML for this message.

As a user, I would like to control the diameter of the outer ring, the 
mechanical advantage, and the size/shape of the interior shaft hole 
(round vs D shaped, for example; maybe spline; or maybe this is done 
outside of this facility).

Specifying the number of planetary gears sounds good because more gears 
allows it to handle more power, but increases friction; I'm not sure I 
would ever deviate from 4.  In fact, if you print 4 and remove one, it 
works just fine.  Setting "N" to 20 would create real problems with 
interference.  I'm not sure that specifying "N" is helpful

I  have no interest in specifying the number of teeth, just the dimensions.

Hope this helps.

Jon

adrianVmariano, 2023-06-30:
Disabling HTML worked.  You can also log in to github and add your post there instead of using the email gateway.  

If you want 3 planetary gears why wouldn't you set N=3 instead of using N=4?  If you print-in-place you can't change the location of the gears later.  I've printed two things with gears, and one of them was print-in place with herringbone gears that lock everything together.  

Setting  N=20 is only a problem if your ratio is more extreme.  If the ratio is close to one they can fit.  :)  I don't know why you would actually want to do that.  (Example below has ratio=0.75 in my code.)  

No interest in the number of teeth?  So if it came out to 73,000 teeth each a micron wide that would be fine?  Or if the sun gear had 3 or 4 teeth?  It seems like there has to be some way to control the scale of the teeth.  I actually wonder a bit about the general way the gears code works, where you specify the tooth size and tooth count.  I get that you need to give tooth size so gears can mesh together, but it seems maybe not entirely natural.   So maybe it would make sense to specify the outer diameter, the width of the ring gear, and a minimum tooth size?  Either a min tooth size or a max tooth count.  The two are interchangeable if I've specified the size of the ring.  

The only way to do this that makes sense is to have a function that computes gear parameters and then you make the gears (with shafts holes as desired) on your own.  Otherwise you ultimately don't have enough control over the gear configuration.   

[image]

[image]

adrianVmariano, 2023-10-20:
Jon, I need your thoughts on the planetary gear implementation.  You said you wanted something that would let you specify the outer diameter.  That is not a well-posed measurement.  You can make it arbitrarily large by thickening the ring gear.  Also, there is no situation where you specify gears by their diameter anywhere in the code.  

Basically a planetary gear assembly is defined by the teeth counts, and it can be any size: just change the gear module as desired.  

```
// Usage:
//   gear_data = planetary_gears(circ_pitch, n, max_teeth, ring_carrier=|carrier_ring=|sun_carrier=|carrier_sun=|sun_ring=|ring_sun=, [helical=], [gear_spin=]);
// Description:
//   Calculates a planetary gear assembly that approximates a desired transmission ratio.  A planetary gear assembly can be regarded as having three
//   elements: the outer ring gear, the central sun gear, and a carrier that holds several planet gears, which fit between the sun and ring
//   The transmission ratio of a planetary gear assembly depends on which element is fixed and which ones are considered the input and output shafts.
//   The fixed element can be the ring gear, the sun gear, or the carrier, and then you specify the desired ratio between the other two.
//   You must also specify a maximum number of teeth on the ring gear.  The function calculates the best approximation to your desired
//   transmission ratio under that constraint:  allowing more teeth will generally give a more accurate approximation.  Note that the planet gears
//   appear uniformly spaced around the sun gear, but this uniformity is often only approximate.  Exact uniformity occurs when teeth_sun+teeth_ring
//   is a multiple of the number of planet gears.
//   .
//   You specify the dedsired ratio using one of six parameters that identify which ratio you want to specify, and which is the driven element.
//   Each different ratio is limited to certain bounds.  For the case of the fixed carrier system, the run and ring rotate in opposite directions.
//   This is sometimes indicated by a negative transmission ratio.  For these cases you can give a positive or negative value.  
//   .
//   The return is a list of entries that describe the elements of the planetary assembly.  The list entries are:
//   - ["sun", teeth, profile_shift, spin]
//   - ["ring", teeth, profile_shift, spin]
//   - ["planets", teeth, profile_shift, spins, positions, angles]
//   The sun and ring gear are assumed to be placed at the origin.  The planet gears are placed at the list of positions.  The gears all
//   have a spin in degrees.  The planets list also includes the angular position of each planet in the `angels` list.
//   One of the planets always appears on the X+ axis when `gear_spin` is zero.
//   .
//   The sun gear appears by default with a tooth pointing on the Y+ axis with no spin, so if gear_spin is not used then the sun gear spin will
//   always be zero.  If you set `gear_spin` then the drive gear for the ratio you specified will be rotated by the specified angle and all
//   of the other gears will be rotated appropriately.
// Arguments:
//   circ_pitch = circular pitch of teeth
//   n = Number of planetary gears
//   max_teeth = maximum number of teeth allowed on the ring gear
//   ring_carrier = set ring/carrier ratio to this value in a ring driven system, must be between 1 and 2
//   carrier_ring = set carrier/ring ratio to this value in a carrier driven system, must be between 1/2 and 1
//   sun_carrier = set sun/carrier ratio to this value in a sun driven system, must be larger than 2
//   carrier_sun = set carrier/sun ratio to this value in a carrier driven system, must be smaller than 1/2
//   ring_sun = set ring/sun ratio to this value in a ring driven system, must have absolute value larger than 1
//   sun_ring = set sun/ring ratio to this value in a sun driven system, must have absolute value smaller than 1
//   helical = create gears with specified helical angle.  Default: 0
//   gear_spin = rotate the driven gear by this number of degrees.  Default:0
function planetary_gears(circ_pitch, n, max_teeth, helical=0, ring_carrier, carrier_ring, sun_carrier, carrier_sun, sun_ring, ring_sun, gear_spin=0) =
    let(
        dummy = one_defined([ring_carrier,carrier_ring,sun_carrier,carrier_sun,sun_ring,ring_sun],
                            "ring_carrier,carrier_ring,sun_carrier,carrier_sun,sun_ring,ring_sun"),
        // ratio is between the sun and ring 
        ratio = is_def(ring_carrier) ? assert(is_finite(ring_carrier) && ring_carrier>1 && ring_carrier<2, "ring/carrier ratio must be between 1 and 2")
                                       ring_carrier - 1
              : is_def(carrier_ring) ? assert(is_finite(carrier_ring) && carrier_ring>1/2 && carrier_ring<1, "carrier/ring ratio must be between 1/2 and 1")
                                       1/carrier_ring - 1
              : is_def(sun_carrier) ?  assert(is_finite(sun_carrier) && sun_carrier>2, "sun/carrier ratio must be larger than 2")
                                       1/(sun_carrier-1)
              : is_def(carrier_sun) ?  assert(is_finite(carrier_sun) && carrier_sun<1/2, "carrier/sun ratio must be smaller than 1/2")
                                       1/(1/carrier_sun-1)
              : is_def(sun_ring) ?     assert(is_finite(sun_ring) && abs(sun_ring)<1, "abs(sun/ring) ratio must be smaller than 1")
                                       abs(sun_ring)
[... cut ...]
ring_gear2d(circ_pitch=pitch, teeth=gear_data[1][1], profile_shift=gear_data[1][2], helical=helical, gear_spin=gear_data[1][3],backing=2);
move_copies(gear_data[2][4])
    spur_gear2d(circ_pitch=pitch, teeth=gear_data[2][1], profile_shift=gear_data[2][2], helical=helical, gear_spin=gear_data[2][3][$idx]);
```

So there's some code.  The only function the pitch has is to correct the planet gear positions.  The only reason helical is needed is for correct calculation of profile shift.  I'm wondering if I should just have it produce the module 1 information and then user can scale as desired.

jon-bondy, 2023-10-21:
Adrian:

My situation was that I needed to fit the outer gear inside a specific space, which is why I specified the diameter.  I suppose it could be smaller, to some extent, but not larger.  I guess the only other important characteristic was the mechanical advantage, the ratio of the gear diameters (I imagine).  

I had not considered that the user could scale the thing to fit: that makes sense to me

I'm not sure I have more to say.  Did I help at all?

adrianVmariano, 2023-10-21:
It's not a good idea to scale a gear with scale() but you can scale the gear by multiplying its tooth size by a scale factor.  I will make this option clear in the docs.  I feel like mentioning this generally somewhere might be a good idea, but not sure where would be the right place.  But really, this same matter applies universally across the library, since the size of gears is determined only indirectly by the tooth count.   Note that mechanical advantage is determined by the ratio of the gears diameters.  You do keep dismissing it, but I'm pretty sure that tooth count also matters, in that 50 micron teeth are not going to be practical.

jon-bondy, 2023-10-22:
If this is a library to create 3D printable objects, it might be worth issuing a warning if the tooth size becomes too small.

adrianVmariano, 2023-10-22:
There is no way to know what "too small" means for the user.   OpenSCAD is dimensionless, so the units are whatever the user chooses.  And printing limits differ between different printing devices and technologies.

closed 2023-10-26 by revarbat