BelfrySCAD/BOSL2 #1193: Add double enveloping worms (aka cone drive)

adrianVmariano, 2023-06-20:
https://conedrive.com/products/hphpa-3-2/gear-sets-materials-specials-etc/

We have curvature on the worm gear to "envelop" the worm but not curvature on the worm to surround the worm gear.

revarbat, 2023-08-09:
This is a first pass attempt:

```openscad
    dbl_env_worm_wall = function (circ_pitch, gear_teeth, starts=1, d, arc, u)
        let(
            helical = asin(starts * circ_pitch / PI / d),
            pr = pitch_radius(circ_pitch, teeth, helical=helical),
            adendum = _adendum(circ_pitch, profile_shift=0),
            rteeth = quantup(ceil(teeth*arc/360),2)+2*starts,
            rack_path = select(rack2d(circ_pitch, rteeth, rounding=false, spin=-90),1,-2),
            m1 = yscale(360/(circ_pitch*teeth)) *
                left(adendum) *
                back(circ_pitch*starts*u),
            taper_table = [
                [-180, 0],
                [-arc/2, 0],
                [-arc/2*0.85, 0.75],
                [-arc/2*0.8, 0.93],
                [-arc/2*0.75, 1],
                [+arc/2*0.75, 1],
                [+arc/2*0.8, 0.93],
                [+arc/2*0.85, 0.75],
                [+arc/2, 0],
                [+180, 0],
            ],
            tarc = 360 / gear_teeth,
            polars = [
                for (p=apply(m1, rack_path))
                if(p.y>=-arc-tarc && p.y<=arc+tarc)
                [pr+p.x*lookup(p.y,taper_table)+adendum, p.y]
            ],
            rpolars = mirror([-1,1],p=polars)
        ) rpolars;

    d=15; arc=60; circ_pitch=5; teeth=40; starts=4;
    hsteps = segs(d/2);
    vsteps = 400;
    helical = asin(starts * circ_pitch / PI / d);
    pr = pitch_radius(circ_pitch, teeth, helical=helical);
    rows = [
        for (i = [0:1:hsteps-1]) let(
            u = i / hsteps,
            theta = (1-u) * 360,
            rpolars = dbl_env_worm_wall(circ_pitch, teeth, starts=starts, d=d, arc=arc, u=u)
        ) [
            for (j = [0:1:vsteps-1]) let(
                v = j / (vsteps-1),
                phi = (v-0.5) * arc,
                minor_r = lookup(phi, rpolars),
                xy = [d/2+pr,0] + polar_to_xy(minor_r,180-phi),
                xyz = xrot(90,p=point3d(xy))
            ) zrot(theta, p=xyz)
        ]
    ];
    vnf = vnf_vertex_array(transpose(rows), col_wrap=true, caps=true);
    vnf_polyhedron(vnf, convexity=teeth);
```
[image]

revarbat, 2023-08-09:
Second take:
```openscad
module dbl_env_worm(circ_pitch, teeth, starts=1, d, arc=60) {
    hsteps = segs(d/2);
    vsteps = 400;
    helical = asin(starts * circ_pitch / PI / d);
    pr = pitch_radius(circ_pitch, teeth, helical=helical);
    adendum = _adendum(circ_pitch, profile_shift=0);
    taper_table = [
        [-180, 0],
        [-arc/2, 0],
        [-arc/2*0.85, 0.75],
        [-arc/2*0.8, 0.93],
        [-arc/2*0.75, 1],
        [+arc/2*0.75, 1],
        [+arc/2*0.8, 0.93],
        [+arc/2*0.85, 0.75],
        [+arc/2, 0],
        [+180, 0],
    ];
    tarc = 360 / teeth;
    rteeth = quantup(ceil(teeth*arc/360),2)+2*starts;
    rack_path = select(rack2d(circ_pitch, rteeth, spin=-90),1,-2);
    m1 = yscale(360/(circ_pitch*teeth)) * left(adendum);
    rows = [
        for (i = [0:1:hsteps-1]) let(
            u = i / hsteps,
            theta = (1-u) * 360,
            m2 = back(circ_pitch*starts*u),
            polars = [
                for (p=apply(m1*m2, rack_path))
                if(p.y>=-arc-tarc && p.y<=arc+tarc)
                [pr+p.x*lookup(p.y,taper_table)+adendum, p.y]
            ],
            rpolars = mirror([-1,1],p=polars)
        ) [
            for (j = [0:1:vsteps-1]) let(
                v = j / (vsteps-1),
                phi = (v-0.5) * arc,
                minor_r = lookup(phi, rpolars),
                xy = [d/2+pr,0] + polar_to_xy(minor_r,180-phi),
                xyz = xrot(90,p=point3d(xy))
            ) zrot(theta, p=xyz)
        ]
    ];
    vnf = vnf_vertex_array(transpose(rows), col_wrap=true, caps=true);
    vnf_polyhedron(vnf, convexity=teeth);
}
```

revarbat, 2023-08-10:
Take three, with internal tooth forms:

```openscad
function dbl_env_worm(
    circ_pitch,
    teeth,
    starts=1,
    d,
    arc=60,
    gear_spin=0,
    anchor=CTR,
    diam_pitch,
    mod,
    pitch,
    spin=0,
    orient=UP
) = let(
        circ_pitch = circular_pitch(circ_pitch=circ_pitch, diam_pitch=diam_pitch, pitch=pitch, mod=mod),
        hsteps = segs(d/2),
        vsteps = 400,
        helical = asin(starts * circ_pitch / PI / d),
        pr = pitch_radius(circ_pitch, teeth, helical=helical),
        adendum = _adendum(circ_pitch, profile_shift=0),
        taper_table = [
            [-180, 0],
            [-arc/2, 0],
            [-arc/2*0.85, 0.75],
            [-arc/2*0.8, 0.93],
            [-arc/2*0.75, 1],
            [+arc/2*0.75, 1],
            [+arc/2*0.8, 0.93],
            [+arc/2*0.85, 0.75],
            [+arc/2, 0],
            [+180, 0],
        ],
        tarc = 360 / teeth,
        rteeth = quantup(ceil(teeth*arc/360),2)+1+2*starts,
        rack_path = select(rack2d(circ_pitch, rteeth, spin=90),1,-2),
        m1 = yscale(360/(circ_pitch*teeth)) * left(adendum),
        rows = [
            for (i = [0:1:hsteps-1]) let(
                u = i / hsteps,
                theta = (1-u) * 360,
                m2 = back(circ_pitch*starts*u),
                polars = [
                    for (p=apply(m1*m2, rack_path))
                    if(p.y>=-arc-tarc && p.y<=arc+tarc)
                    [pr+p.x*lookup(p.y,taper_table)+adendum, p.y]
                ],
                rpolars = mirror([-1,1],p=polars)
            ) [
                for (j = [0:1:vsteps-1]) let(
                    v = j / (vsteps-1),
                    phi = (v-0.5) * arc,
                    minor_r = lookup(phi, rpolars),
                    xy = [d/2+pr,0] + polar_to_xy(minor_r,180-phi),
                    xyz = xrot(90,p=point3d(xy))
                ) zrot(theta, p=xyz)
            ]
        ],
        ys = column(flatten(rows),1),
        miny = min(ys),
        maxy = max(ys),
        vnf1 = vnf_vertex_array(transpose(rows), col_wrap=true, caps=true),
        vnf = zrot(gear_spin, p=vnf1)
    ) reorient(anchor,spin,orient, d=d, l=maxy-miny, p=vnf);

module dbl_env_worm(
    circ_pitch,
    teeth,
    starts=1,
    d,
    arc=60,
    gear_spin=0,
    diam_pitch,
    mod,
    pitch,
    anchor=CTR,
    spin=0,
    orient=UP
) {
    vnf = dbl_env_worm(
        teeth=teeth,
        starts=starts,
        d=d,
        arc=arc,
        gear_spin=gear_spin,
        circ_pitch=circ_pitch,
        diam_pitch=diam_pitch,
        mod=mod,
        pitch=pitch
    );
    bounds = pointlist_bounds(vnf[0]);
    delta = bounds[1] - bounds[0];
    attachable(anchor,spin,orient, d=max(delta.x,delta.y), l=delta.z) {
        vnf_polyhedron(vnf, convexity=teeth);
        children();
    }
}
```

revarbat, 2023-08-10:
Still need to correctly calculate `hsteps=`

revarbat, 2023-08-10:
Current form:
[image]

PR #1250 (2023-08-11): Added double_enveloping_worm()
Resolves #1193
closed 2023-08-11 by revarbat