BelfrySCAD/BOSL2 #1196: [BUG]

SnowballsChance, 2023-06-27:
**Describe the bug**
It looks like the code inside a diff is running 3 times

**Code To Reproduce Bug**
```

include <BOSL2/std.scad>
echo("step 1");
cube(10, center = true);

diff()
{
    echo("step 2");
    left(15)
    cube(10, center = true);

    echo("step 3");
    left(17) up(5) fwd(3)
    tag("remove") cube(10, center = true);

}

echo("step 4");

difference()
{
    echo("step 5");
    left(15)
    cube(10, center = true);

    echo("step 6");
    left(17) up(5) fwd(3)
    cube(10, center = true);

}

echo("step 7");

.
```

**Expected behaviour**
I expected the that each echo statement to only be printed once, the ones in the BOSL2 diff repeat 3 times, the standard difference behaves as expected.

Does this mean the geometry exists 3 times?

**Screenshots**
Log below

```

Compiling design (CSG Tree generation)...
ECHO: "step 1"
ECHO: "step 2"
ECHO: "step 3"
ECHO: "step 2"
ECHO: "step 3"
ECHO: "step 2"
ECHO: "step 3"
ECHO: "step 4"
ECHO: "step 5"
ECHO: "step 6"
ECHO: "step 7"
Compiling design (CSG Products generation)...
Geometries in cache: 2
Geometry cache size in bytes: 1456
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized tree has 5 elements!
Compile and preview finished.
Total rendering time: 0:00:00.035
```

**Additional context**

- OpenSCAD Version:
-  version 2023.06.12 (git 48f159d73) 

- Other libraries used:
-

closed 2023-06-27 by revarbat

SnowballsChance, 2023-06-27:
Thanks I was worried I was in some sort of Star Trek Time loop thing.
Really appreciate the great work that has gone into BOSL2.

bendavis78, 2023-09-23:
> It runs once to get the geometry to subtract from, once to get the geometry to difference away, and once for the geometry that you want kept despite the difference.

This seems less like "expected behavior" and more "consequence of implementation". With a simple + child diff operation, a typical user would expect each geometry to be calculated once. Instead we get 3 iterations per geometry (1 parent + 1 child = 6 iterations).

This can add up and lead to severe performance issues. Is there no room for optimization?

adrianVmariano, 2024-01-16:
One could perhaps imagine OpenSCAD being redesigned to make a more efficient implementation possible.  Feel free to push for that, but I think it would require a radical change, where information could flow from children to parents instead of only parents to children.