BelfrySCAD/BOSL2 #1200: [BUG]

jon-bondy, 2023-07-02:
**Describe the bug**

A difference() fails between two linear_sweep()s.  It worked before I added the scale parameter

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>

inches = 25.4;

sw = 1.8 * inches;            // snoot width
sl = 7 * inches;            // snoot length
th = 2;
ss = 0.2;                    // snoot scale

    difference() {
        tex = $preview? undef : texture("rough", roughness=0.5);
        linear_sweep(regular_ngon(n=3, d = sw + 4*th), h=sl, convexity = 10,
scale = ss);
        translate([0, 0, -1])
            linear_sweep(regular_ngon(n=3, d = sw), texture=tex, h=sl + 2,
tex_size=[10,10], style="min_edge", convexity = 10, scale = ss);
        }
```

**Expected behavior**
The difference() works with F5 and fails with F6

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 2023.06.24 (git 92eec2d43)
- BOSL version: from 6/28/23
- Other libraries used:

adrianVmariano, 2023-07-02:
Looks like linear_sweep never produces a valid polyhedron when you give it a texture.  Or at least it very often fails.  I tried several textures and none worked.  Reduced code example:

```
include <BOSL2/std.scad>

inches = 25.4;

sw = 1.8 * inches;            // snoot width
sl = 7 * inches;            // snoot length
th = 2;
ss = 0.2;                    // snoot scale

tex = texture("diamonds");
linear_sweep(regular_ngon(n=3, d = sw), texture=tex, h=sl + 2,
             tex_size=[10,10], style="min_edge", convexity = 10, scale = 1);
cube(20);
```

jon-bondy, 2023-07-19:
Please let me know when/if this is fixed.

closed 2023-08-04 by revarbat