BelfrySCAD/BOSL2 #1203: Including BOSL2 in a library breaks "use" of that library unless BOSL2 is also included in the top file

pipatron, 2023-07-14:
I got some confusing errors when I split my larger project into smaller parts. Turns out that including BOSL2 in a library file that you then `use` from your main project breaks some standard modules:

My OpenSCAD parts file `mylib.scad`:
```
include <BOSL2/std.scad>
width = 1;   // A common name I don't want to export
function testfun() = width;
```

Now I want to use it in `main.scad`:
```
use <mylib.scad>  // "use" because I don't want to export variables or run code
echo(testfun());
cube();
```

Instead of a nice cube I got an assertion error:
```
ECHO: 1
WARNING: Ignoring unknown variable '$anchor_override' in file BOSL2/attachments.scad, line 2494
WARNING: Ignoring unknown variable '$anchor_override' in file BOSL2/attachments.scad, line 2496
WARNING: Ignoring unknown variable '$attach_to' in file BOSL2/attachments.scad, line 3100
WARNING: Ignoring unknown variable '$tags_shown' in file BOSL2/attachments.scad, line 3471
WARNING: Ignoring unknown variable '$tags_shown' in file BOSL2/attachments.scad, line 3471
ERROR: Assertion '(is_list($tags_shown) || ($tags_shown == "ALL"))' failed in file BOSL2/attachments.scad, line 3471
TRACE: called by '_is_shown' in file BOSL2/attachments.scad, line 2518
TRACE: called by 'if' in file BOSL2/attachments.scad, line 2518
TRACE: called by 'multmatrix' in file BOSL2/attachments.scad, line 2510
TRACE: call of 'attachable(anchor = [-1, -1, -1], spin = 0, orient = [0, 0, 1], size = [1, 1, 1], size2 = undef, shift = undef, r = undef, r1 = undef, r2 = undef, d = undef, d1 = undef, d2 = undef, l = undef, h = undef, vnf = undef, path = undef, region = undef, extent = true, cp = [0, 0, 0], offset = [0, 0, 0], anchors = [], two_d = false, axis = [0, 0, 1], override = undef, geom = ["prismoid", [1, 1, 1], [1, 1], [0, 0], [0, 0, 1], [0, 0, 0], [0, 0, 0], []])' in file BOSL2/attachments.scad, line 2488
TRACE: called by 'attachable' in file BOSL2/shapes3d.scad, line 62
TRACE: call of 'cube(size = [1, 1, 1], center = undef, anchor = [-1, -1, -1], spin = 0, orient = [0, 0, 1])' in file BOSL2/shapes3d.scad, line 58
TRACE: called by 'cube' in file main.scad, line 3
Current top level object is empty.
```

Including BOSL2 in `main.scad` solves the problem, as does replacing `use <mylib.scad>` with `include <mylib.scad>`.

I don't know if this is possible to solve using the current OpenSCAD include/use feature, and it is of no real importance in any of my current projects, but it _would_ be nice to be able to use BOSL2 internally in my personal libraries without having to do workarounds such as splitting my libraries into multiple levels of includes, etc.

revarbat, 2023-07-14:
`use` is just plain broken in OpenSCAD and is not recommended.

pipatron, 2023-07-14:
Ok! I'm certainly not going to argue that point, having very little experience with cases like it. I just haven't seen anyone warn against it before, so I assumed `use` was perfectly fine for data hiding and writing "clean" libraries in general. I'll stick with `include` for now.

adrianVmariano, 2023-07-14:
`use` causes the top level assignments to be executed again every time any function in the file is called.  You can get thousands of unnecessary executions this way, resulting in code that runs slowly.

closed 2023-08-01 by adrianVmariano