BelfrySCAD/BOSL2 #1204: documentation clarification for threading/screws

jon-bondy, 2023-07-14:
I am trying to roll my own threads, taking measurements from an existing object.  As such, I am constrained to measure between the tips of the threads.  When one specifies a diameter in BOSL for a screw, is that the diameter including the threads, or not including the threads?  And how can one calculate the depth of the threads?

Thank you

Jon

revarbat, 2023-07-14:
Threads are included in the diameter.

Standard ISO threads are 60° triangles with the top 1/8th and bottom 1/4th of their height truncated. 

https://en.m.wikipedia.org/wiki/ISO_metric_screw_thread

adrianVmariano, 2023-07-14:
If he's taking measurements from something, the threads are presumably non-standard.  If they are not nonstandard, he should be using a thread gauge.  Or align the measurements with a standard.    He doesn't say what the thread geometry is.

jon-bondy, 2023-07-14:
Thank you!   Here is the code I am working with.  Lines 28 and 35 are the key.  Measurements are on lines 14 and 15; line 13 is a guess.

// external/male M25.4x0.75 (32 tpi?)
// internal/female 28.5 mm 1.125" 42 tpi

include <BOSL2/std.scad>
include <BOSL2/threading.scad>

inches = 25.4;
eps = 0.01;
$fn = 200;

// ID on the left at the tips of the threads is 26.55 (0.3 deep)
// OD on the right at the tips of the threads is 25.30 (0.3 deep?)
td = 0.4;			// thread depth
lid = 26.50;
rod = 25.40;
lir = lid/2;
ror = rod/2;

th = 2;
od = lid + 2*th;	// OD of entire adapter
or = od/2;

etd = rod;	// external thread diameter

module Try3D2nd() {
	// right side - bottom
	difference() {
		threaded_rod(d = rod - 2*td, l = 4, pitch = 0.75);
		translate([0, 0, -eps])
			cylinder(h = 4 + 4*eps, d = rod - 2*td - 2*th, center = true);
		}
	// left side - top
	translate([0, 0, 7])
		intersection() {
			threaded_nut(nutwidth = lid + 2*th, id = lid, h = 14, pitch = 0.605);
			cylinder(h = 10, d = lid + 2*th, center = true);
			}
	// connector - middle
	translate([0, 0, 2])
		difference() {
			cylinder(h = th, d = lid + 2*td);
			translate([0, 0, -eps])
				cylinder(h = th + 2*eps, d2 = lid, d1 = rod - 2*td - 2*th);
			}
	}
!difference() {	
Try3D2nd();
*cube(100);
}

adrianVmariano, 2023-07-14:
I think this isn't really a BOSL2 specific issue, except for the matter of documentation.  The docs currently state: "You can give the outer diameter..." and then in the arg list it says "d = Outer diameter of threaded rod".  This seems to me to be fairly clear about the meaning of the parameter.  What doc improvement do you suggest?  

Regarding your specific issue, you might want to ask on a generic forum, because as I said, it's not really a library issue.  It looks like your object has a 1 inch OD.  You should work in inches!  Note that INCH is defined in BOSL2.  So you can write threaded_rod(d=INCH,...).  If the object is in inches then it's likely that the thread pitch is specified in threads per inch, so trying to measure that way would be sensible.  

But you haven't said anything about the object.  If it's plastic or glass it may use SPI threads (see bottlecaps.scad), though the SP style caps have diameters in nominal metric measurements and nothing as fine as 32 tpi.  Certainly if you have some random object with nonstandard threading there's no guarantee that it uses ISO style 60 deg threads, so using threaded_rod may not give the correct thread form.   It could be ACME, for example.  That's might be somewhat tricky to assess by looking at a part in hand, though I guess maybe with ACME you could see flats on the tops of the threads.  Your very tiny threads might be difficult to 3d print, if that's you're plan.

jon-bondy, 2023-07-15:
I agree this is not BOSL2-specific.

If the diameter between the tips of the threads is the diameter stated in the threaded rod module, that works; but you cannot use a similar approach when you create the nut.  That would make the threads have zero height.  I guess what you wrote is clear enough.  My situation is rare and bizarre.

I was given the spec, as strange as it might be.  Now that I have an actual sample part here, I can measure with a caliper, and the dimensions are not as stated in the spec.  Not off by much, but by enough.

I am making a custom adapter for use with a telescope.  I printed one with my resin printer (Elegoo Saturn 2) and the threads are perfect, and they match the mating parts precisely.

Thanks for your help!

Jon

adrianVmariano, 2023-07-15:
Can you clarify what you mean about the nut?  You're trying to make the nut by subtracting the threaded rod from something----not using trapzeoidal_threaded_nut---right?  So the outer diameter of the object produced by threaded_rod is the value you specify.  If you then subtract this from something that becomes the diameter to the thread bottoms in the nut side.  You will need to increase the outer diameter to create clearance in the nut (or decrease it on the screw part).  Is there something I should say in the manual somewhere to clarify things?  

Oooh, you have a resin printer.  I imagine 42 tpi is no problem at all---though I don't know anything about strength of the parts from SLA.

jon-bondy, 2023-07-15:
I used threaded_nut() because it was easier for me to use than subtracting a threaded rod from a larger cylinder.  Maybe not a good idea, but that is what I did.

Had I subtracted a threaded rod from a larger cylinder, I need to decrease the specified diameter if I want to get the tips of the "nut" threads to have a specified diameter.  And I do not know the depth of the threads, except by approximation and guess work.

Perhaps an example where someone rolls their own threaded rod and "nut" would be helpful.

[image]

adrianVmariano, 2023-07-15:
The "id" parameter for threaded_nut is the od of the rod that gets subtracted, not the tooth-to-tooth minimum id.  It seems like this is not clear in the manual.   I'm not sure of the best way to document this.  

I wonder if there should be a "round" option for the shape parameter of threaded_nut.

jon-bondy, 2023-07-15:
If the "od" or "d" parameter for the rod goes to the tip of the threads, and the "id" parameter for the nut is NOT to the tip of the threads, then something needs to be said.  The best way to document this may be with a simple example.

The threaded_nut() facility is an easy way to create an internally threaded rod, so adding a "round" option could be useful.

Thanks for all of your help!

adrianVmariano, 2023-07-15:
I don't see how an example clarifies anything.  You can't tell the difference between fractions of a mm in a example preview.  I need to say something that describes what the id parameter for a nut means, but I'm not sure of the most clear way to do that.  It seems like actually the parameter is misnamed and should be called d_maj or something like that.  I could write it as "id specificies the major diameter of the threads" but I don't know if people will know what that means.   Or "id specifies the major diameter of the threads (the outer diameter of the rod that is subtracted to make the threads)".   Or "id specifies the diameter of the threads from root to root (the major diameter)".  

I could add a similar explanation to the rod making module, that "od specifies the diameter from crest to crest (the major diameter)".  

Does that seem clear?

jon-bondy, 2023-07-16:
The example image might not help, but the example code might be helpful.  Something like

diam = 10;
td = 0.4;              // thread depth
threaded_rod(d = diam);
nut(d = diam - td);

"the outer diameter for threaded rods is measured across the tips of the threads (and is larger than the solid part of the rod), while inner diameter for threaded nuts is measured at the bottom of the threads (where the nut is solid and not threaded)" or something like that.

Is that helpful?

adrianVmariano, 2023-07-16:
Well, certainly you do seem to be still somewhat confused based on your example code.  

It should be something like

`threaded_rod(d=diam,length=20)`

to mate with

`threaded_nut(nutwidth=diam+5, id=diam, $slop=.15);`

or alternatively if you want to create clearance explicitly

`threaded_nut(nutwidth=diam+5,id=diam+.6);`

The adjustment to the diameter of the threaded_nut has nothing to do with thread depth.  It's a clearance gap so that the hole in the nut is bigger than the rod.  Without that, you're subtracting the rod to make the nut and so the rod is the exact same size as the hole and hence won't fit.  Is it possible you're overcomplicating this in your head?  

It's precisely analogous to doing:

cyl(d=10);

difference(){
  cuboid(20);
  cyl(d=10+.6, h=100);
}

where when I make the hole, I need it a bit oversized.  The two cyl objects would otherwise have the same diameter, and even with perfect fabrication accuracy, the rod would not fit in the hole.  

If this concept needs to be illustrated by examples to be clear then where do these examples go?  Do I need to add this same example 12 times to cover the six threaded_rod modules and the six threaded nut modules?

jon-bondy, 2023-07-16:
"Is it possible you're overcomplicating this in your head?"  Almost certainly!

This code would be sufficient for me to understand it.  Perhaps it is already present. 

threaded_rod(d=diam,length=20)
to mate with
threaded_nut(nutwidth=diam+5, id=diam, $slop=.15);

The situation I was faced with was a somewhat suspect specification (like nominal diameters that were clearly wrong) so I ended up having to measure the ID and OD of the threaded surfaces.  And that meant that while my measurement was spot on for the threaded rod OD, it was off by the thread depth for the threaded nut ID.  So I had to figure out empirically how much to adjust the thread diameter.  I could not use the method you used in the above code.

Maybe no one else will run into this situation.  Perhaps we have spent as much time on this, or more, than is warranted.

adrianVmariano, 2023-07-16:
Actually in your situation, you measure the od of the threaded rod surface.  That gives you the threaded rod outer thread size.  The nut inner diameter needs to be that same measured value increased by a clearance.  You could hope that the difference between your two measurements gives you the thread depth, another parameter you'd like to know, but the fact that there are clearance gaps makes this somewhat prone to error---your measurement is going to actually give you the thread depth plus the clearance gap.  The thing is, the clearance isn't a critical parameter, but if you have the thread depth wrong it changes the thread geometry, which would mean that the threads don't mate properly, which I imagine could shorten the lifespan of the threaded part.  

I think that at the very least, clarifying exactly what diameter is intended is worth doing.  I got the impression that you didn't find "root to root" to be a clear explanation, though.   I could write something about it being the diameter of the threaded rod that is subtracted.

adrianVmariano, 2023-07-26:
Added clarification to docs.

closed 2023-07-26 by adrianVmariano