BelfrySCAD/BOSL2 #1212: round_corners() of partially round shape?

hoditohod, 2023-07-24:
First: incredible library! Moves OpenSCAD to a new level!

I'm not sure if this is the right space to post questions. I haven't found any reference to a forum/irc/discord on the landing page, if one exists please put it in the toplevel README.md!

Question:
I would like to round a shape that is roughly half of a circle:

    include <BOSL2/std.scad>
    include <BOSL2/rounding.scad>
    s = right_half(circle(d=15, $fn=30), x=1);
    polygon(round_corners(s, r=1));

While this works for $fn=30, I would like to have it smoother and it stops working for larger values. There's an error stating "Roundovers are to big...", but honestly I don't know how to proceed. 

Any help appreciated!

hoditohod, 2023-07-25:
After going through the doc a few times, I ended up with this (derived from path_join example):

    include <BOSL2/std.scad>
    include <BOSL2/rounding.scad>
    arc1 = arc(width=20, thickness=8,$fn=75);
    horiz = [[-10,0],[10,0]];
    stroke(path_join([arc1,horiz], 2,k=1,closed=true), width=.3,closed=true,$fn=12);

closed 2023-07-25 by hoditohod

adrianVmariano, 2023-07-25:
Could use path_smooth on a roughly shaped polygon to produce all the curves.