BelfrySCAD/BOSL2 #1256: Minor documentation issue(s)[BUG]

jon-bondy, 2023-08-18:
In 

https://github.com/BelfrySCAD/BOSL2/wiki/Tutorial-Attachments

search for typo ""singlke "

I am confused by the following: in the first example below the typo we find "align([RIGHT+TOP,LEFT+TOP])" which creates two instances of the prismoid.  Fair enough.  But if we skip down three examples, we find that "align(RIGHT+TOP,DOWN)" creates ONE instance rather than two.  I thought the comma in the 1st example caused the creation of multiple instances, so why does the comma in the 2nd example not do so?

RAMilewski, 2023-08-18:
Not a bug.   Note the square brackets [ ] in the first example, that don't appear in the second.   I suspect align([RIGHT+TOP,DOWN]) produces the result you were expecting:

[image]

revarbat, 2023-08-19:
PR checks are working.

PR #1257 (2023-08-19): Typo fix in Attachments tutorial.
Fixes #1256
closed 2023-08-19 by revarbat