BelfrySCAD/BOSL2 #1270: Mesh not closed with textures[BUG]

jon-bondy, 2023-09-02:
**Describe the bug**
When I try to apply textures to a surface F6 fails to complete.  Flush caches prior to F6

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>

inches = 25.4;
eps = 0.01;
slop = 0.5;

// camera
ch = 40.86 + slop;	// height
cd = 26.05 + slop;	// depth (X)
cw = 84 + slop;		// width (Y)
cc = 40;			// cable
cmo = 6.5;			// mount offset

th = 2;

// snoot
sl = 7 * inches;			// snoot length
sw = 1.8 * inches;			// snoot width
sa = 21;					// snoot angle
so = sw/2 * sin(30) - 0.47;	// snoot offset
ss = 0.2;						// snoot scale - 0.2
smd = 22 + slop;			// slit mount diameter

gpx = 0.05*sl;
gpy = 0.22*cw;

module OuterSnoot()
	linear_sweep(regular_ngon(n=3, d = sw + 4*th), h=sl, convexity = 10, scale = ss);	

module SnootSideProfile()
	projection()
		rotate([90, 0, 0]) 
			OuterSnoot();
*SnootSideProfile();

module SnootShadow()
	projection()
		rotate([0, -90, 0]) 
			OuterSnoot();

module SnootSupport()	
	// support under the now-raised snoot
	difference() {
		translate([-13.3, 0, 0])
			rotate([0, 90, 0]) 
				linear_extrude(0.23 *sw)
					SnootShadow();
		translate([0, -30, 0])
			rotate([-90, 0, 0]) 
				linear_extrude(100, convexity = 10)
					SnootSideProfile();
		}

module RoughSnoot() {
	difference() {
		tex = $preview? undef : texture("rough", roughness=0.5);
//		tex = undef;
		OuterSnoot();
		translate([0, 0, -1])
			linear_sweep(regular_ngon(n=3, d = sw), texture=tex, h=sl + 2, tex_size=[10,10], style="min_edge", convexity = 10, scale = ss);	
		}
	SnootSupport();
	
*	difference() {
		translate([-sw, -th/2, 0])
			cube([sw, th, sl]);
		OuterSnoot();
		translate([-1000 - so - 2.5, -500, 0])
			cube(1000);
		}
	// integrated slit
	translate([0, 0, sl - th/2])
		scale(0.92)
			difference() {
				st = 0.5;
				// slit surface
				linear_sweep(regular_ngon(n=3, d = ss*(sw + 7*th)), h=1, convexity = 10);	
				// slit
				translate([-0.2*sw*ss, -st/2, -eps])				
					cube([ 0.6*sw*ss, st, th + 2*eps]);
				}
	}	

module Snoot() {
	// gobal positioning
	difference() {
		translate([gpx, gpy, 0])
			rotate([0, 0, sa])

		translate([0, 0, so + 2.5])
			rotate([0, -90, 0]) 
				RoughSnoot();
		translate([0, -100, -eps])
			cube(1000);	
		}
	}

Snoot();

```

**Expected behavior**
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron.
Geometries in cache: 24
Geometry cache size in bytes: 95732088
CGAL Polyhedrons in cache: 11
CGAL cache size in bytes: 351176
Total rendering time: 0:01:03.665
Top level object is a 3D object:
   Simple:     yes
   Vertices:       33
   Halfedges:     104
   Edges:          52
   Halffacets:     46
   Facets:         23
   Volumes:         2
Rendering finished.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: [OpenSCAD](https://www.openscad.org/) version 2023.06.24 (git 92eec2d43)
- Other libraries used:

RAMilewski, 2023-09-15:
Renders ok for me on OpenSCAD 2023.08.21 (git 7322a5df7).   I suspect this may an OpenSCAD issue not a BOSL2 issue.

jon-bondy, 2023-09-16:
I installed 2023.09.10 and F6 completes without an error in the console, BUT the resulting object is solid rather than hollow.  If I do F5, I can see the correct hollow object (albeit without the texture).

adrianVmariano, 2023-10-04:
I tried this in 2021.01 and it renders for me.  It did take 75 minutes.   I agree that it's looking unclear if this is a BOSL2 issue.  This uses a random texture.  I wonder if some instantiations are invalid and others are not.  (That would be a BOSL2 issue.)  This leads to the question of: can you get it to fail with a fixed random seed and preferably with a smaller model size.  

[image]

adrianVmariano, 2023-10-04:
I noticed also that the texture is apparently too deep for the model, as it pokes through the front (back?) end:

[image]

jon-bondy, 2023-10-04:
I added these two lines near the top:

// try to set the random number generator seed to a known value
single_rand = rands(0,10,1)[0];

It failed in the usual way with a solid interior.  Does this help at all?

[OpenSCAD](https://www.openscad.org/) version 2023.09.10 (git dcaf25f0e)

Parsing design (AST generation)...
Saved backup file: C:/Users/jon/Documents/OpenSCAD/backups/Wooden Bowl Display Structure-backup-WZrmracl.scad
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron.
Geometries in cache: 24
Geometry cache size in bytes: 95732088
CGAL Polyhedrons in cache: 11
CGAL cache size in bytes: 351176
Total rendering time: 0:01:02.379
Top level object is a 3D object:
   Simple:     yes
   Vertices:       33
   Halfedges:     104
   Edges:          52
   Halffacets:     46
   Facets:         23
   Volumes:         2
Rendering finished.

adrianVmariano, 2023-10-04:
That doesn't actually set the random seed.  You need 

     a = rands(0,1,1,seed=444);

at the top or something like that.  With the above I have failure under 2021.01 and 2023.9.15 under Linux.

jon-bondy, 2023-10-05:
This (failure) is what I get with the above code:

Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron.
Geometries in cache: 24
Geometry cache size in bytes: 95732088
CGAL Polyhedrons in cache: 11
CGAL cache size in bytes: 351176
Total rendering time: 0:01:01.203
Top level object is a 3D object:
   Simple:     yes
   Vertices:       33
   Halfedges:     104
   Edges:          52
   Halffacets:     46
   Facets:         23
   Volumes:         2
Rendering finished.

adrianVmariano, 2023-10-05:
Note that the key thing we need here is for Revar to get it to fail, since he's the one who knows how the texture code works.  And he reports that under the 2022 version he's running, it's not failing.  The details of the fail message don't tell us anything.  There are a lot of different messages that can appear depending on CGAL internals and they all mean "polyhedron was bad".   When you say it "completes" and you get a solid object, I still see that there is a CGAL error, hence failure.  It's just not including the textured object, hence the solid output.  

Note that the triangular endcap isn't sized correctly and leaves a hairline gap around its edge, which is visible even without texture, but creates a irregular textured gap.  I eliminated this from the model but the model still fails, so while it looked questionable, it wasn't the cause of the problem.   Actually I went ahead and did a full reduction.  Really when providing a bug report the bug reporter should do this.  Result:

```
include <BOSL2/std.scad>

a=rands(0,1,1,seed=444);

inches = 25.4;
th = 2;

// snoot
sl = 3 * inches;			// snoot length
sw = 1.8 * inches;			// snoot width
ss = 0.2;						// snoot scale - 0.2

module RoughSnoot() {
        tex = texture("rough", roughness=0.5);
	linear_sweep(regular_ngon(n=3, d = sw), texture=tex, h=sl + 2, tex_size=[20,20], style="min_edge", convexity = 10, scale = ss);	
        cube(2);
}

RoughSnoot();
```

And the interesting thing is that with Thrown Together turned on, I get this:

[image]

which suggests that the problem is a simple face reversal of the textured section.  If that's all that's going on it should be easy for Revar to fix it.  I tried changing texture to "diamonds" and got this:

[image]

So the problem isn't about the rough texture specifically.   Note also how this analysis is much easier to do when you get rid of the extraneous parts of the model.

jon-bondy, 2023-10-05:
Thank you for simplifying the sample code.  Let's hope that it is as simple as face reversal!

revarbat, 2023-10-05:
... I just tried that simplified code, and Thrown Together mode shows no purple at all.  In either 2021.01 or 2022.08.22 (git ea8950bd3)

revarbat, 2023-10-05:
[image]

revarbat, 2023-10-05:
Same for diamonds texture.

jon-bondy, 2023-10-05:
My take, from the above, is that Adrian and Revar are seeing different results from the same code.  I am not sure what I can contribute here, other than concern that such differences are occurring.

adrianVmariano, 2023-10-06:
Arg.  It turns out my non-development version of BOSL2 (that gets loaded if you just say include<BOSL2/std.scad>) was kinda old.  I updated it and my purple faces disappeared.  And under 2023.9.15, render works:

[image]

With my features set like this:

[image]

If I turn on "manifold" then it fails but without an error.  I get this:

[image]

I don't feel like re-running in 2021.01.  I think on the basis of my previous test on a different computer it works there.   I think the conclusion is that this is an OpenSCAD bug in the "manifold" rendering engine.  It should probably be made into a minimal example and posted to the OpenSCAD github as a bug.   My previous reduced example is too reduced: it does not trigger the problem.

jon-bondy, 2023-10-07:
I set my features as recommended, above, and generated a correct surface two times in a row, so perhaps this issue is resolved.

Thank you

adrianVmariano, 2023-10-07:
So really this needs to be posted as an OpenSCAD bug, because it's possible that down the road the "manifold" method will replace the other methods and become the algorithm that OpenSCAD uses to do renders.  If that were to happen, then you'd be in trouble because there'd be no way to render your model.  I believe that right now, the manifold method is the fastest render method in the dev build.   I normally turn it on

jon-bondy, 2023-10-07:
Would you do that, please.  I fear that I will not state the bug correctly or clearly enough.

adrianVmariano, 2023-10-07:
I tried to make a minimal example.  What's actually happening is that a zero thickness layer is being left on the front face after the cube subtraction.  

```
include <BOSL2/std.scad>

a=rands(0,1,1,seed=444);

inches = 25.4;
eps = 0.01;
slop = 0.5;

// camera
ch = 40.86 + slop;	// height
cd = 26.05 + slop;	// depth (X)
cw = 84 + slop;		// width (Y)
cc = 40;			// cable
cmo = 6.5;			// mount offset

th = 2;

// snoot
sl = 7 * inches;		// snoot length
sw = 1.8 * inches;		// snoot width
sa = 21;			// snoot angle
so = sw/2 * sin(30) - 0.47;	// snoot offset
ss = .2;//0.2;			// snoot scale - 0.2

gpx = 0.05*sl;
gpy = 0.22*cw;

module OuterSnoot()
	linear_sweep(regular_ngon(n=3, d = sw + 4*th), h=sl, convexity = 10, scale = ss);	

module RoughSnoot() {
	difference() {
		tex = texture("rough", roughness=0.5);
		OuterSnoot();
		translate([0, 0, -1])
			linear_sweep(regular_ngon(n=3, d = sw), texture=tex, h=sl + 2, tex_size=[10,10], style="min_edge", convexity = 10, scale = ss);	
		}
	}	

module Snoot() {
	difference() {
		translate([gpx, gpy, 0])
			rotate([0, 0, sa])
                        translate([0, 0, so + 2.5])
			rotate([0, -90, 0]) 
				RoughSnoot();
		translate([0, -100, -eps])
			cube(1000);	
		}
}

difference(){
  Snoot();
  back(10)cuboid(1000,anchor=FWD);
}
```

produces:

[image]

with manifold turned on.  And actually without the extra cube subtraction you see a solid front face in this model, but if you look in the skinny end of the snoot it's correct.

adrianVmariano, 2023-10-07:
Making ss some other value seems to "fix" it.   Also getting rid of the cutting cube in Snoot() fixes it.  And so does changing the cube to size 200.  Apparently "manifold" doesn't like the big cube?

adrianVmariano, 2023-10-07:
I posted this issue:  https://github.com/openscad/openscad/issues/4769 Anybody with anything to add can put it there.  And I think we can close out this BOSL2 issue.

revarbat, 2023-10-12:
This is definitely a Manifold bug, and one that they already have a tenative fix for. ( https://github.com/elalish/manifold/pull/564 )

Closing this.

closed 2023-10-12 by revarbat