BelfrySCAD/BOSL2 #1277: Offline documentation

pfelecan, 2023-09-25:
How can I generate the documentation of the project offline ?
There are situations when I'm working offline and need to access the documentation.
There are also some issues when GitHub is slow or non responsive.
Finally, sometimes I'm not using the head of the tree and the documentation advanced in regard with the version that I'm using.
Thank you in advance of the generous soul who can give me the recipe to this endeavor.

RAMilewski, 2023-09-25:
It's not a pretty as the wiki, but the documentation is actually just comments in the .scad files.  @revarbat may be able to suggest how to generate the HTML files locally using his [docsgen](https://github.com/revarbat/openscad_docsgen) utility.  I tried that once and failed, but I didn't spend much time trying.

revarbat, 2023-09-26:
To just download the BOSL2 wiki locally, so you can view it offline, you can do the following:

1. Install a multi-file aware MarkDown viewer/editor.  On Mac, I recommend Typora.
2. Clone the GitHub BOSL2 wiki repo locally. `git clone https://github.com/BelfrySCAD/BOSL2.wiki.git`
3. Open the `BOSL2.wiki/Home.md` file with your MarkDown viewer.

To generate a local copy of the docs from sources, follow these steps:

1. Install a MarkDown viewer/editor.  On Mac, I recommend Typora.
2. Install a recent Python version.
4. Install `pip` and `setuptools` if needed.  Recent Python versions don't need this step.
5. `pip install openscad_docsgen`
6. Clone the GitHub BOSL2 repo locally, if you haven't already. `git clone https://github.com/BelfrySCAD/BOSL2.git`
7. `cd BOSL2`
8. Clone the GitHub BOSL2.wiki repo locally. `git clone https://github.com/BelfrySCAD/BOSL2.wiki.git`
9. Run docs generation in the BOSL2 directory: `openscad-docsgen`
10. Wait a few minutes.
11. Run tutorial generation in the BOSL2 directory: `openscad-mdimggen`
12. Open `BOSL2.wiki/Home.md` with your MarkDown viewer/editor.

To nuke the docs from orbit, and regenerate EVERYTHING:

1. Install a MarkDown viewer/editor.  On Mac, I recommend Typora.
2. Install a recent Python version.
4. Install `pip` and `setuptools` if needed.  Recent Python versions don't need this step.
5. `pip install openscad_docsgen`
6. Clone the GitHub BOSL2 repo locally, if you haven't already. `git clone https://github.com/BelfrySCAD/BOSL2.git`
7. `cd BOSL2`
8. Clone the GitHub BOSL2.wiki repo locally. `git clone https://github.com/BelfrySCAD/BOSL2.wiki.git`
9. `cd BOSL2.wiki`
10. `rm -rf images *.scad.md`
11. Run docs generation in the BOSL2 directory: `openscad-docsgen`
12. Wait half an hour or more.
13. Run tutorial generation in the BOSL2 directory: `openscad-mdimggen`
14. Open `BOSL2.wiki/Home.md` with your MarkDown viewer/editor.

revarbat, 2023-09-26:
I have just generated a 2138 page 64MB hyperlinked PDF file of all the docs, but it's too big to attach to this Issue.
To generate it, you need to install pandoc, and run the ./scripts/mkdocspdf.sh script after generating or downloading the BOSL2.wiki docs as shown above.  Then you need to open the 160MB BOSL2_Docs.html file, and print the result to PDF.  I've had success doing this using Safari on the Mac.