BelfrySCAD/BOSL2 #1284: Gear Attachments

adrianVmariano, 2023-10-05:
There was a plan to implement gear attachments that would position gears and align the teeth.

malv-c, 2025-03-19:
my problem should be simpler
ie:

$fn=128;ang=12.5;m=1.4;g=71;e=22;
module w(v1,v2,v3,S,s){bevel_gear(mod=m,v1,v2,v3,spiral=0,backing=5,shaft_diam=S
    ,anchor="apex")
  cyl(h=48,d=3attach=BACK
color("lightblue")
xrot(ang)
  bevel_gear(mod=m,v2,v1,v3,spiral=0,right_handed=true,shaft_diam=s,anchor="apex")
    cyl(h=165,d=3,anchor=BACK,orient=UP);}
    
    w(g,e,ang,85,21);
i don't find a way to attach cyl to gear base  or top

adrianVmariano, 2025-03-19:
Is your problem about attaching gears to other gears so the teeth mesh?  If so there's no simple way to do that until this feature is implemented, which will not be soon.  If it's something else, open a new issue and post an actual working example with a better explanation of what you are trying to do.

malv-c, 2025-03-19:
using spin=(teeth number/meshing teeth number) should be a good start
anyway here is  [Gear Attachments] so i did think a problem of Gear Attachment cylinder related was on topic....

closed 2026-10-30 by adrianVmariano