BelfrySCAD/BOSL2 #1298: Add chamfer and rounding to the top of `text3d()` result

DeflateAwning, 2023-10-28:
**Is your feature request related to a problem? Please describe.**
I wish I could create text that has a rounded or chamfered top.

**Describe the solution you'd like**
I wish I could pass a `rounding` arg to `text3d`, like most other `shapes3d()` things.

**Describe alternatives you've considered**
I'm open to suggestions. Haven't found a way at all though.

**Example Code**
```
// how it is now
text3d("Hello World", anchor=BOTTOM, h=20, size=8);

// how I wish it worked
text3d("Hello World", anchor=BOTTOM, h=20, size=8, rounding=2, except=[BOTTOM]);
```

RAMilewski, 2023-10-28:
I've battled this problem as well.  It would be a great help when making cookie cutters with text. (Something friends and familly ask for often).

revarbat, 2023-10-28:
Sadly, we have no access to the font geometry data to be able to do this, and an offset-based variant of linear_extrude() doesn't yet exist in OpenSCAD. 

HOWEVER, there is a convoluted way to do what you want:

1. Create some text in Inkscape with the font you want. 
2. Convert the text to paths. 
3. Save an SVG of it. 
4. Open the SVG file in a text editor. 
5. Take the <path d="..."> strings from that file put it in a .scad file. 
6. In OpenSCAD, use the Pathbuilder library to convert those strings to pointlist paths.
7.  Put the paths together in a list to make a Region. 
8. Use BOSL2's `offset_sweep()` from rounding.scad to get your rounded result.

revarbat, 2023-10-28:
Actually, I am unsure if `offset_sweep()` can handle Regions, so it is possible you will have to treat the region subpaths individually and do some creative difference.

adrianVmariano, 2023-10-31:
offset_sweep requires a path.  I suppose in principle the module version only could be extended to do regions.   Not sure if there are hidden complications there.  (There might be, associated with flipping the direction of the end treatments for subtracted components.)  

Just in case this is unclear, the original request of this issue is impossible.  OpenSCAD doesn't give us access to the geometry of the text.

DeflateAwning, 2023-10-31:
How can we request that this be updated in the OpenSCAD system then? Seems like a massive oversight.

adrianVmariano, 2023-10-31:
The inability to extract geometry from OpenSCAD is a huge limitation in many ways, not just for text. There is development under way to address this, but who knows if or when it will appear in the main release.

adrianVmariano, 2023-11-04:
Are we keeping this open for some reason?

closed 2023-11-04 by revarbat

drothmaler, 2025-07-11:
just for the next person coming around this issue: you can use `minkowski` + a `cylinder`/`shere` to add a chamfer/rounding to a 3d text; but be aware, that this will take a ***REALLY*** long time to render (depending on the `$fn` used).