BelfrySCAD/BOSL2 #1310: Unable to access older versions of BOSL2

peachsteak, 2023-11-06:
Hey,

I'm using the following tool in OpenSCAD - https://github.com/charmaur/PolyDiceGenerator
I had this installed on an old pc and all was running smoothly however on my new machine I'm running into issues which seem to be due to the BOSL2 version it's looking for (X/Y placements of fonts going crazy).

The code requires BOSL 2.0.525 but I can't see any versioning on your repo. Is there a way to access this so that I can use the tool again?

alastaira, 2024-07-22:
The version number is defined in version.scad. So, if you want v2.0.525, browse to the commit of the repository that updated the value to BOSL_VERSION = [2,0,525]; 

Which is... https://github.com/BelfrySCAD/BOSL2/tree/a1755d0a370cae558bca74199c9d6d99422e82b8

xloem, 2024-12-18:
This issue can be closed.

closed 2025-04-06 by adrianVmariano