BelfrySCAD/BOSL2 #1312: [BUG] Objects built with BOSL2 hull() and union() have errors that prevent offset() from working

amatulic, 2023-11-14:
**Bug description**
Sometimes, attempting to perform `offset()` on a path of coordinate points that was built by multiple `hull()` operations combined together by `union()` result in one of two errors I have observed:
* ERROR: Assertion '!approx(segment[0], segment[1])' failed: "Path has repeated points" in file..."
* ERROR: Assertion '!approx(segment[0], segment[1])' failed: "Path has repeated points" in file..."

**Code To Reproduce Bug**
The first error I managed to reduce to a concise amount of code to a reasonable size. I have not yet been able to do that for the second error, but both are generated from the same module in my .scad script.

The code below generates two ovals in different ways, and the ovals are colinear and share an end point.
It crashes with `$fn=16`. But change `$fn` to 17 and it works.
```
include <BOSL2/std.scad>
include <BOSL2/rounding.scad>

dx = 12; // circle offset
$fn=16; // 16 generates an error, 17 works fine

blank_cookie();

module blank_cookie() {
    // oval made by hulling two circles
    oval_circ = hull_polygons([
        move([-dx,0], p=circle(d=12)),
        move([dx,0], p=circle(d=12))
    ]);
    // long skinny rectangle to be offset into an oval
    bone = move([dx,-0.02], p=square([24,0.04]));
    // here is the oval
    oval_rect = offset(r=6, bone, closed=true);
    // combine the two ovals
    combined = union(oval_circ, oval_rect)[0]; //[0] gets rid of region nesting
    // it crashes here if $fn=16
    roundoutline = offset(r=-2, offset(delta=2, combined, closed=true), closed=true);
    polygon(points=roundoutline);
}

// hull together an array of polygons paths
// p is the form [ [[x,y],[x,y],...], [[x,y],[x,y],...],... ]
function hull_polygons(p) = let(
    pts = flatten(p), // big list of coordinates
    paths = hull2d_path(pts), // vertex indices for hull
    n = len(paths)-1) [
        for(i=[0:n]) pts[paths[i]]
    ];

// flatten an array of arrays (from OpenSCAD documentation)
function flatten(l) = [ for (a = l) for (b = a) b ] ;
```

**Additional context**
* Using OpenSCAD 2021.01
* Only BOSL2 libraries are used

It seems that `hull()` might occasionally get some vertices out of order. It also seems that `offset()` can result in a vertex that is a 180° turn.

adrianVmariano, 2023-11-14:
If you have an example where hull() gets vertices out of order you should post a specific bug report for that.  Note that debug_polygon() might be helpful.  I don't think it's possible for offset() to create a 180 deg turn because that's the specific case that generates the "parallel check failed" message.  But creating such a result is only a bug if it's not the right answer.   You might want to use select() and the flatten() provided by BOSL2.  

I examined your first example.  Because you  union one object that is 12 units tall with another one 12.04 units tall there will necessarily be close points in the union.  This is not a bug.  Really I wouldn't recommend using the function forms of union and offset to do what you're doing here.  The function forms of union and offset are very complicated, slow, and likely to misbehave in edge cases.  I think it would actually be easier to construct your model by concatenating arcs.  Or the model as you have shown it could be done without point lists using the native module forms of hull, union and offset.  
So what goes wrong in this case is that the tiny segment that joints the slightly different width portions ends up being flagged as an invalid segment in the offset.  I don't know how to make offset() more robust to this, other than possibly a more sophisticated algorithm that is an order of magnitude slower---I decided that the trade off there isn't worth the cost.  

[image]

The above image shows the situation.  Your path is above in black.  It has a tiny segment where the two parts of your model with different widths meet.  At the bottom you can see the raw offset segments.  The little segment is inside the yellow region, which is the offset of the original shape (as computed by the native offset module).  Because that segment is inside that region it is treated as invalid and removed from the output.  The other two segments are parallel, and don't have an intersection, so they invoke the parallelcheck error.  

You can work around this problem if you really want to offset shapes like this by setting `checkvalid=false` though that might get you in trouble in other situations.  The docs for offset() mention this in the second paragraph, that this test can fail, though the error reporting certainly doesn't make clear what has happened.  

Regarding repeated points, it does seem that offset is creating repeated points.  I'll have to investigate further to know what's going on, but in many cases this is the **desired** behavior for offset, so it's not necessarily a bug.  You may just need to run `deduplicate()`.  

I'm not sure what your goal is, but your second example makes it seem like you might consider using stroke() or offset_stroke() to generate your model.

amatulic, 2023-11-14:
Hi @adrianVmariano - thanks for that analysis. I now understand why `offset()` identified parallel segments. However, that doesn't explain why a simple array of polygon vertices generates the same error.

Since I reported this yesterday, I've done some major modifications and rearrangements to my initial report, so you may have missed the two-line code example (now at the top of the report) that produces the 180° turn error. It wasn't clear from your reply if you tested that. It's just a simple polygon on which `offset()` is applied. It fails due to the points that are close together. They don't make a 180° turn as the error message states. Here's the code:
````
include <BOSL2/std.scad>
vertices =  [
 [-18, 0], [-17.5433, -2.2961], [-16.2426, -4.24264], [-14.2961, -5.54328], [-12, -6],
 [11.8995, -6], [12, -6.02], [36, -6.02], [38.2961, -5.56328], [40.2426, -4.26264],
 [41.5433, -2.3161], [42, -0.02], [42, 0.02], [41.5433, 2.3161], [40.2426, 4.26264],
 [38.2961, 5.56328], [36, 6.02], [12, 6.02], [11.8995, 6],
 [-12, 6], [-14.2961, 5.54328], [-16.2426, 4.24264], [-17.5433, 2.2961]];
expanded = offset(delta=2, vertices, closed=true); // this crashes
echo(expanded); // we never get here
````
> I think it would actually be easier to construct your model by concatenating arcs.

For the simple examples I provided here, yes. I've done that in other projects. For this project, it wouldn't be easier.

> Regarding repeated points, it does seem that offset is creating repeated points

No, it's `union()` that creates repeated points in this case. This happens before `offset()` is called, which aborts due to the repeated points in the input path. I verified this by echoing the array after the hull operation, and it did generate two of the same point, corresponding to the location where two shapes happen to have a vertex in the same place. `union()` is the source of the problem for this error.

I know it isn't obvious from the code examples why I'm doing things the way I'm doing. The code was extracted from a much larger and more complex script that works most of the time, but not all of the time due to these errors that crop up. (I have some parameters being generated at random to test things, and I can get an error less than 5% of the time). For this project I need to manipulate vertex paths. If I do it without BOSL2, I can still get the same 2D polygon in the end, but then I need to extrude it and round the top perimeter. The only way I can see to do that is to use `offset_sweep()`, but that (as far as I can tell) requires a path of vertices.

adrianVmariano, 2023-11-14:
Yes, offset_sweep requires a path of vertices.  (In fact, that's the sort of application for which you need offset to duplicate points.)  

When I examined your second example, you have shape=union(..) and that shape output does not contain duplicate points because its length is the same as the length of running deduplicate on it.  But expanded=offset(...) does contain duplicate points, because deduplicate shrinks its length.  The error doesn't hit until you run offset on the previous offset output.  
(I also double checked and the shortest segment in shape, the output of union, is 0.16.  There are no duplicate points there.)  

I think it's probably better if you add new comments instead of editing your original one, because that makes it hard to tell if/when you have changed something there.

amatulic, 2023-11-14:
> When I examined your second example, you have shape=union(..) and that shape output does not contain duplicate points

You are correct. I was confused by my own example.

adrianVmariano, 2023-11-14:
I took a look at your example that you added to the top.  It's the same example as in the longer code, just pulled out.  So my previous explanation should explain why "a simple array of polygon vertices generates the same error."  It's the same example I plotted above, where there is a small segment that ends up failing the validity check and the two adjacent segments are parallel, so they don't have an intersection.  (It's the lack of intersection that triggers the "parallel" warning.)   And yes, it's not a reversal like the message says.  I didn't know it was possible to have two segments parallel that didn't share an endpoint without it being a reversal.  It's a little unclear how to write a clear error message for this.  Note that you could simplify the sample down to four points and it would still fail.  

Using deduplicate with eps=0.3 seems risky unless you're sure you're not going to lose other important points, but if that's not a concern it would resolve the issue.   It really does look like you're trying to reimplement offset_stroke, so if you haven't considered that, give it a look.

amatulic, 2023-11-14:
OK, `dedupllicate()` fixed the problem for me with duplicated points.

However, there is still an issue with that first short bit of code, doing an offset of a simple path with no duplicates, just with two vertices very close together. I still get the 180° turn error. Even if I run it through `deduplicate()` with `eps=0.1`, it removes only one of the two pairs of vertices that are too close together.

amatulic, 2023-11-14:
Also, why is that polygon being treated as if it had two parallel unconnected segments? Wouldn't the short little distance between the ends of those two segments count as another segment? It's supposed to be a closed path, after all.

adrianVmariano, 2023-11-14:
Checking your example with path_segments_lengths reveals that the bad segments have length 0.1025, so you could remove them with that epsilon.  You could use `check_valid=false` if that doesn't cause problems elsewhere, of you can stop unioning together things that disagree by .02 so that you don't get those bad segments.   

The diagram with the unconnected segments shows what happens in the middle of the offset() calculation.  Each segment gets offset by the specified offset amount.  That disconnects the segments, or it may overlap them.  The segments get extended or trimmed to the intersection point.  But the hard part of the offset process is getting rid of the invalid parts.  The code assumes a segment is invalid if it's closer to the original curve than the offset distance---in that case it would be contained inside the offset.  For your example this test fails and mistakenly removes the short segment.  That's why disabling the validity check would fix it----but as possible cost of breaking other parts of your offset, depending on what's happening there.

amatulic, 2023-11-15:
Ah, thank you. I see now. Deduplicating the near values should fix it in this case, I can see how that might mess up outward offsets into a concave curve.

amatulic, 2023-11-15:
I also see what you mean about the dangers of using `check_valid=false`. When I try to offset a union of a 16-sided polygon and a square, I get little notches in the offset polygon when `check_valid=false`, but it looks correct when `check_valid=true'.
````
include <BOSL2/std.scad>

$fn = 16;
shape = union(circle(d=30), move([0,-15], p=square(10.8, center=true)))[0];
expanded = offset(delta=2, shape, closed=true, check_valid=false);
polygon(points=expanded);
color("blue") stroke(shape, closed=true, width=0.2);
````
Results in this:
[image]

I think I need to just live with the occasional error I'm getting. De-duplicating fixes some of them. Twiddling with input parameters or the $fn value can fix some of the others.

adrianVmariano, 2023-11-15:
Yeah, those validity checks are there for a reason.  Sometimes you can get away with skipping them (which will make offset() much faster to run since the checks are O(N^2)) but not generally.  

You don't seem to be paying attention to my suggestion to look at offset_stroke and/or to avoid using slightly mismatched size elements.  If the objects you union weren't slightly different sizes that would also eliminate the problematic little segments.   Your current code uses basically a clumsy method to make the object where you offset a teeny thin square that is oversized.  Why are you doing this?  Is there something essential in your model that it needs to be .02 wider on the right side?  Offset a line instead of a square and make it exact.  You can do it manually, where you offset the shape and add arc() at each end...or use offset_stroke:

`oval_rect = offset_stroke([[dx,0],[dx+24,0]], width=12,end="round",start="round");`

I'm a little puzzled about why this isn't the best solution.

amatulic, 2023-11-16:
Hi, I was delayed today due to a November 15 tax filing deadline for nonprofit orgs (I'm the treasurer so I have to do it).

I looked at `offset_stroke()` but didn't use it for these reasons:
* It returns two polygons, for the inside perimeter and outside perimeter, and I need only the outside one. It seems like a waste, but it isn't a hindrance.
* most importantly, `offset_stroke()` calls `offset()`, which is what's giving me problems in the first place, so I don't know how it would help. Wouldn't I just get the same errors? Or is it doing something different?

You suggested avoiding using slightly mismatched size elements. Usually they aren't. When `offset()` crashes my script, I examine what happened, and so far, the underlying cause is always vertices close together or duplicated. I don't deliberately do this. The 2D outline I'm making is a humanoid figure constructed from the union of several other shapes, with parameters controlling their size, position, and orientation. There is a "randomize" feature in my script to output a figure in a random pose. Every once in a while, a combination of these random parameters results in an error with `offset()`.

I can live with this, but because it happened more than once, I created create this issue here. Clearly I didn't understand what was going on until you explained it.

I made some changes to my script and I can no longer reproduce the errors with the parameters that were causing them. I need to keep generating random ones until I get an error, and then I'll try offset_stroke.

amatulic, 2023-11-16:
Many randomly generated tests later, I got the direction reveral error. Using `offset_stroke()` in place of `offset()` generated the same error, as I suspected it would.

Again, this is happening rarely. If it isn't worth fixing, this issue can be closed as "working as intended".

adrianVmariano, 2023-11-16:
My suggestion for offset_stroke is that you would use it one time, because it produces a shape that looks like what you get at the end of your current process, where you are calling offset several times, e.g. to do rounding.   Note that offset_stroke returns two paths only if the in put you give it is closed.  You would presumably give it an open path.  You could for example have a path that does left leg-body-left arm and then another one that does right leg body right arm and you could union those together.  (Yes, body is duplicated.)   The human figure would still require the union of multiple objects, so ultimately this helps mainly if you can avoid running offset on outputs of offset or union because the rounding is already done.

Another thought about this is that I don't know if you need the humanoid figure as a path.  Your process appears to be to generate parts, union them together, and then round using the double offset call.  On the example code, you could do the double offset call using the native module form of offset, which will not have this problem.   That is, you could use the round2d() module to do that final rounding step. 

I have been wondering if I could make the offset function more robust by testing for validity after computing intersections between adjacent segments instead of before.

amatulic, 2023-11-16:
Yes, the double offset is for creating rounded corners both convex and concave. Using the native offset isn't feasible. The native offset would prevent me from using `offset_sweep()` to get a rounded perimeter of the final 3D shape. There is no way to do that with native OpenSCAD calls without using `minkowski()` unless I create many extrusions of different heights and offset to simulate a rounded perimeter with microscopic stair-steps. I felt that was a rather clunky approach that creates far too many facets.

adrianVmariano, 2023-11-16:
```

path = [
        [0,0],
        [5,5],
        [10,0],
        [5,5],
        [10,6],
        [5,5],
        [5,10],
        [5,5],
        [-1,4],
        [5,5]
        ];

op=offset(offset(path, r=1.5,closed=true),r=-.75,closed=true);

stroke([op],width=.1);
```

I wonder if the above helps you.  I have implemented 180 deg turns.  I tried a stick figure model where it traces out the shape with overlaps and then uses two offsets to get the rounding. I think that this will never encounter the "short segment" problem.   

This code isn't posted yet.  There are issues with the chamfer=true case.  But the result looks like this:

[image]

Another strategy just occurred to me, which would be to use rectangular body parts and then round_corners, but that might fail sometimes if corners aren't always where they are expected.    Unfortunately the idea I had for making offset() more robust was a bad one.  Probably that's why I didn't do it that way to begin with.  It fails catastrophically in many cases.

amatulic, 2023-11-16:
Wow, those are pretty severe 180° turns. I've wanted something like that for other projects I've done, in which I had to create a curved path with rounded ends by making a curve that doubles back on itself with a tiny thickness to avoid a 180° turn at the ends, and then offset that polygon. With that, I could get a perfect rounded end without needing to add that tiny thickness.

That should definitely help here too, especially since the errors I encountered didn't actually involve 180° turns, just a microscopically short segment separating two parallel long segments.

I got rid of one offset. Everything I'm trying to union together is made up of primitive rounded shapes. Therefore, instead of doing a double offset, I simply generated those shapes with bigger dimensions without changing their locations. Circles get a bigger diameter, squares get bigger dimensions, they are combined with `hull()` as needed. Then after combining everything with `union()`, I use a negative `offset()` only once to shrink the shape to make the concave corners curved. This eliminates the need for the initial expansion offset, and further reduces the error rate.

Errors still happen, but I have to work harder to get them, including fiddling with $fn to cause two vertices from the original components to be close together.

adrianVmariano, 2023-11-16:
A 180 deg turn is a 180 deg turn.  I've always made rounded objects with round_corners.  I wrote offset for offset_sweep, really.  It occurs to me that you need your starting shape to be well-designed for offset if you want it to go into offset_sweep.   In any case, I believe the above strategy will always produce a uniformly spaced layout of points at the curves and no small segments, but I don't know enough about your project to understand how much that helps over all.

adrianVmariano, 2023-11-18:
The update has been posted with support of 180 deg turns.

adrianVmariano, 2023-11-21:
I'm closing this issue now as "working as intended" as you suggested above.  I think the mods it inspired are helpful: true 180 deg support in offset() and better error reporting.  The underlying issue that short segments in paths given to offset cause problem is one I don't have any idea how to solve.

closed 2023-11-21 by adrianVmariano

amatulic, 2023-11-21:
@adrianVmariano - thank you. So far, I haven't encountered any aborts due to errors with this new release.

However, I have encountered a weirdness with this:
````
include <BOSL2/std.scad>

shape =  [
[-1.14, -45.95],
[-1.5, -44.15], 
[4.3, -35.5],
[16.54, -40.56],
[16.89, -42.34], // change 16.89 to 16.9 for a different result
[17.1, -42.3],
[22.3, -68.5],
[4.3, -73.8],
[-1.3, -46]
];

oshape = offset(r=-3, shape, closed=true);
polygon(points=oshape);
````
[image]

Deleting any one of those vertices removes one or both extensions. The extension shown on the lower right is from a small stair step on the upper left, and the extension on the upper side is coming from a stair step on the upper right of the shape.

adrianVmariano, 2023-11-21:
It's the same problem: the little segments are getting erroneously removed and then the remaining segments are almost parallel so their intersection is way off toward infinity.  If they were *actually* parallel you'd get an error.  I didn't check in detail, but probably the slight perturbation that fixes it is causing the segment to get retained instead of deleted.