BelfrySCAD/BOSL2 #1313: thread_helix output fails CGAL

adrianVmariano, 2023-11-14:
```
include <BOSL2/std.scad>
include <BOSL2/bottlecaps.scad>
sp_neck(48,400,2);
```

The above code fails CGAL because vnf_vertex_array produces a polyhedron that CGAL objects to.  It seems to be necessary to clear the cache before the error appears, even when running this code the first time.  vnf_validate reports only "BIG FACE" errors for the two endcaps.

revarbat, 2023-11-28:
I was able to fix this by running the VNF through `vnf_triangulate()` in `spiral_sweep()`.  This indicates to me that the BIG FACEs are not planar enough for CGAL's precision.  Soooo, FP rounding errors issue. It still previews in 0.186s on my laptop, so this should be an acceptable solution.

adrianVmariano, 2023-11-28:
Erhm, I tested coplanarity of the big faces and it was passing is_coplanar(), so that means they are coplanar to 1e-9.  

If it's a problem here, won't it be a problem elsewhere?  That is, don't we need to triangulate the caps in vnf_vertex_array rather than doing it in spiral_sweep()?

revarbat, 2023-11-28:
As I recall, we made an explicit decision to not do a possibly-expensive `vnf_triangulate()` in `vnf_vertex_array()`, but to keep `vnf_triangulate()` available for the user to invoke in the cases that do give CGAL problems.

revarbat, 2023-11-28:
Mind you, I wouldn't mind a `triangulate=true` argument to invoke triangulation.

adrianVmariano, 2023-11-28:
How slow is it to triangulate the caps?  Arguably we can triangulate **just** the caps since we know the other faces are all triangles already.  The problem I see is that none of us even knew how to fix this problem, so it's pretty non-obvious.  Seems like if we leave it as-is we should add the triangulate=true option and document this.

commit 84a302b5 (2023-11-28): Triangulate big faces in spiral_sweep().  Fixes #1313
closed 2023-11-29 by revarbat