BelfrySCAD/BOSL2 #1317: In place hinge

MikiGrit, 2023-11-27:
**Is your feature request related to a problem? Please describe.**
I would like to be able to create in-place hinge. The obvious use-case is 3D printing.

**Describe the solution you'd like**
I imagine something like this: https://www.thingiverse.com/thing:2187167 I hope it would not be difficult to add new parameter to hinge functions (https://github.com/BelfrySCAD/BOSL2/wiki/hinges.scad) already present in BOSL2 library.

Thanks for consideration.

revarbat, 2023-11-28:
Wouldn't `knuckle_hinge()` ( https://github.com/BelfrySCAD/BOSL2/wiki/hinges.scad#module-knuckle_hinge ) cover this?

MikiGrit, 2023-11-28:
Maybe I'm missing something but I did not find the way how to fill the inner hole with rod that would attach to one side. Kind of what you can see in this image: 
[image]

And yes, I hoped that it could be simple addition of new parameter to the `knuckle_hinge` function whether it should be created with inner filled or not.

Note that both sides should be printed at once. That's why I called it in place printed hinge.

adrianVmariano, 2023-11-30:
There's no support for that in the existing library.  But I wonder, will this reliably print?  You're relying on bridging to accurately build that rod.  Does that work?  The way I've seen in-place hinges done is with cones that nest along the hinge so the angles are always >45, but that would require more than the simple addition of a component to the existing model.

MikiGrit, 2023-11-30:
From my experience, you have to use a little force to get it moving. But after that, it is fine.

And you are right. Hinge with cone would be even better. I just also thought that the cone would be more difficult, so I suggested the rod.