BelfrySCAD/BOSL2 #1320: [BUG] Should rotate_sweep() treat $fn,$fs,$fa the same as rotate_extrude()?

pipatron, 2023-11-29:
I wanted to see exactly how the built-in `rotate_extrude()` dealt with `$fn` when it comes to angles less than 360°, and I noticed that BOSL2 `rotate_sweep()` is different.

`rotate_sweep()` calculates the number of segments over the whole revolution, and then constructs a shape using _all_ of them no matter how small the section is:

    steps = segs(max_x)

...while OpenSCAD built-in scales with the angle:

    fragments = (unsigned int)std::ceil(fmax(Calc::get_fragments_from_r(max_x - min_x, node.fn, node.fs, node.fa) * std::abs(node.angle) / 360, 1));

I think it should scale with the angle, otherwise it adds a lot of unnecessary small polygons, ignoring `$fs` limits etc. Here's a telling example:

```
include <BOSL2/std.scad>
$fn=16;
for(i=[1:4]) {
  a = i/4*90;
  right(i*2) {
    rotate_sweep(right(1,circle(r=1,$fn=4)),angle=a);
    fwd(2)
      rotate_extrude(angle=a)
        right(1) circle(r=1,$fn=4);
  }
}
```
[image]

Top row is BOSL2, bottom row is OpenSCAD.

The only reason I'm not changing this myself and submitting a pull request is that I don't know if it will mess up some of the modules that depend on it. If you think this is a good idea I could give it a try, see if it ruins anything.

adrianVmariano, 2023-11-30:
I think it's a BOSL2 bug.

revarbat, 2023-12-07:
It's definitely a bug.

closed 2023-12-10 by revarbat