BelfrySCAD/BOSL2 #1328: Creating a bowl-shaped patch from a path

s-leroux, 2023-12-11:
I have previous experience in using vanilla OpenSCAD to create very basic parts, but I am new to BOSL2. I read the docs on the wiki; the library is very impressive.

I am restoring a Japy Message typewriter. It does not have much value except a sentimental one. It is a 1963 typewriter, used by my mum in her first job up to the early 70s when I was born. I try to model a key cap as many were damaged by decades of storage.

[image]
[image]

I managed to create the overall shape of the key cap using smooth paths around trapezoids.  However, I have no idea how to model the top of the caps. The frame is planar, but the inside is bowl-shaped. I imagine creating a Bezier patch from the path and shifting down the inner control point. But I can't see how to achieve that. Unless there is a different approach to the problem?

[image]
```
include <BOSL2/std.scad>
include <BOSL2/rounding.scad>

top_width = 13;
bottom_width = 15;
keycap_height = 9;

top_face = smooth_path(
    trapezoid(w1=top_width, w2=top_width-2, h=15.5),
    closed=true);

bottom_face = smooth_path(
    trapezoid(w1=bottom_width, w2=bottom_width-2, h=15.5),
    closed=true);
    
skin(slices=0,
    profiles=[bottom_face, top_face],
    z=[0, keycap_height]);
```

adrianVmariano, 2023-12-12:
I don't have an easy answer to this.  It looks like you don't actually want a bowl.  You want a flat recess with gently sloping sides.  You could do a recess maybe by taking your border shape, offsetting it in, and using offset_sweep to make a version that is flat on the bottom, with a rounded profile.  You then subtract that from your original.

s-leroux, 2023-12-14:
Thanks, Adrian, for the answer. I didn't manage to use offset_sweep() for my purpose. My best solution for now was to create the "finger rest" shape as the hull() around the top face and a sphere (whose radius was empirically estimated). When I flatten the shape using zscale(), it doesn't look that bad. Maybe it is a bit too "conical" compared with the original. I wonder if this will be noticeable for a 1mm-height recess. I will probably print a prototype to validate the design.

```
module finger_rest() {
    r = 7;
    up(keycap_height+1/r/2) {
        zscale(1/r) hull() {
            linear_extrude(height=1, center=true)
                polygon(top_face);
            sphere(r);
        }
    }
}

module key() {  
    skin(slices=0,
        profiles=[bottom_face, top_face],
        z=[0, keycap_height]);
}

difference() {
    key();
    finger_rest();
}
```

adrianVmariano, 2023-12-15:
You probably need to use offset_sweep() to chamfer or round that shape.

adrianVmariano, 2024-01-03:
Does this issue call for any action, or can we close it?

closed 2024-01-08 by adrianVmariano

s-leroux, 2024-01-08:
Thanks a lot for your help with that issue.