BelfrySCAD/BOSL2 #1329: Acme (probably all) nuts are heptagonal. Chamfering problem?

jbash, 2023-12-14:
**Describe the bug**

On today's master branch at commit 8a84e05788319bbfab2b7ef004e984f7bb671b1f,  the code below produces a nut with a heptagonal (7-sided, not *hex*agonal) body. This is with OpenSCAD 2021.01 as shipped with Fedora 39.

```
include <lib/BOSL2/std.scad>
include <lib/BOSL2/threading.scad>
acme_threaded_nut(nutwidth=1.25*INCH, id=0.75*INCH, tpi=5, height=1.25*INCH);
```

Apparently `acme_threaded_nut` uses `trapezoidal_threaded_nut` which uses `generic_threaded_nut` which uses `_nutshape` which calls `cyl` with $fn set to 6. And in fact `cyl` itself gives a (chamfered) heptagonal prism when called like this:

```
include <lib/BOSL2/std.scad>
include <lib/BOSL2/threading.scad>
cyl(d=1.20*INCH, circum=true, $fn=6, l=1.25*INCH, chamfer1=.0120*INCH, chamfer2=.0120*INCH);
```

However, if  the chamfer1 and chamfer2 arguments are removed, the result is a hexagonal prism.

**Expected behavior**

Hexagonal nut.

adrianVmariano, 2023-12-15:
I can't duplicate this.  I ran the test code and got:

[image]

Only six sides, not seven.

jbash, 2023-12-15:
Hmm. I just retried it with a fresh clone, and it still fails for me.

[image]

Presumably it's some interaction with the version of OpenSCAD or one of its dependencies.

If I were you, I'd probably just close this as probably not BOSL's bug and probably not worth working around just to accommodate some unknown and probably uncommon OpenSCAD issue.

More information if you do care to look at it:

The Fedora package starts with a tarball named "openscad-2021.01.src.tar.gz", and applies 6 patches on top of it.

+ Something called "openscad-polyclipping.patch", which looks like it just changes where it looks for some header files, probably to avoid building in a library that the system packages separately.
+ -cgal5.3.patch , claimed to be "CGAL 5.3 fixes from https://github.com/openscad/openscad/pull/3844"
+ -fix-overloaded-join.patch , apparently a backport or cherry pick from OpenSCAD commit 9b79576c1ee9d57d0f4a5de5c1365bb87c548f36
+ -cgal-build-fix.patch , ditto 71f2831c0484c3f35cbf44e1d1dc2c857384100b
+  CVE-2022-0496.patch , 770e3234cbfe66edbc0333f796b46d36a74aa652
+ CVE-2022-0497.patch , 84addf3c1efbd51d8ff424b7da276400bbfa1a4b

Packages providing shared libraries:

+ boost-filesystem-1.81.0-8.fc39.x86_64
+ boost-program-options-1.81.0-8.fc39.x86_64
+ cairo-1.18.0-1.fc39.x86_64
+ double-conversion-3.1.5-9.fc39.x86_64
+ fontconfig-2.14.2-5.fc39.x86_64
+ freetype-2.13.1-2.fc39.x86_64
+ glib2-2.78.1-1.fc39.x86_64
+ glibc-2.38-11.fc39.x86_64
+ gmp-6.2.1-5.fc39.x86_64
+ harfbuzz-8.2.1-2.fc39.x86_64
+ lib3mf-2.2.0-5.fc39.x86_64
+ libGLEW-2.2.0-5.fc39.x86_64
+ libX11-1.8.7-1.fc39.x86_64
+ libgcc-13.2.1-6.fc39.x86_64
+ libglvnd-glx-1.7.0-1.fc39.x86_64
+ libspnav-1.1-2.fc39.x86_64
+ libstdc++-13.2.1-6.fc39.x86_64
+ libxml2-2.10.4-3.fc39.x86_64
+ libzip-1.10.1-1.fc39.x86_64
+ mesa-libGLU-9.0.3-1.fc39.x86_64
+ mpfr-4.2.0-3.fc39.x86_64
+ opencsg-1.5.0-4.fc39.x86_64
+ polyclipping-6.4.2-18.fc39.x86_64
+ qscintilla-qt5-2.13.4-3.fc39.x86_64
+ qt5-qtbase-5.15.11-6.fc39.x86_64
+ qt5-qtbase-gui-5.15.11-6.fc39.x86_64
+ qt5-qtgamepad-5.15.11-1.fc39.x86_64
+ qt5-qtmultimedia-5.15.11-1.fc39.x86_64

adrianVmariano, 2023-12-15:
I swore I did a pull before running the test, but apparently not.  I think @revarbat 's recent fix for the faceting in rotate_sweep() has broken things.  I am now able to duplicate the problem.

revarbat, 2023-12-18:
... I had thought that that fix would only apply to `rotate_sweep()` with angle= or less than 360º.  I'll review my exact changes and fix this.

PR #1331 (2023-12-18): Fix for #1329:  wrong $fn in cyl(), rotate_extrude()
Fixes #1329
closed 2023-12-18 by revarbat