BelfrySCAD/BOSL2 #1330: Improve arc() for easier usage

dan-p3rry, 2023-12-15:
**Is your feature request related to a problem? Please describe.**
When using arc(radius = R, angle = A, wedge = true, spin = S);
more than once, like in a for loop, will not render, because all the arcs share the point [0,0]
 
**Describe the solution you'd like**
Add an option to arc() to offset the arc origin a smidgen from [0,0].  The most accurate implementation would be equivalent to:
difference() {
    arc(r=R, angle = A, wedge = true, spin = S);
    arc(r = 0.1, angle = A, wedge = true, spin = S);
}

**Describe alternatives you've considered**
This works but takes more thinking/effort because p1/p2 have to be calculated.
arc(cp=[0, 0.01], points = [p1, p2], wedge = true, spin = S);

**Example Code**
```
// Code goes here.
```

**Additional context**
Add any other context or screenshots about the feature request here.

adrianVmariano, 2023-12-15:
So really what you're asking for is a ring2d() module, like the 2d equivalent of tube()?

dan-p3rry, 2023-12-16:
That works if it takes angle and spin as arguments.

Dan

revarbat, 2023-12-30:
I don't think I quite understand the original posted problem.  Are you attempting to form a Region from multiple arcs, and it's failing because they all share a vertex at `[0,0]`?  (Oddly, the function version of `arc()` doesn't take a spin argument.) Are you attempting to make a hollowed out wedge?

To make a hollowed wedge arc:
```openscad
R=50; A=72; S=30; W=5;
arcpath = zrot(S, p=arc(r=R, angle=A, wedge=true));
rgn = [ arcpath, offset(arcpath, r=-W, closed=true) ];
region(rgn);
```
[image]

dan-p3rry, 2023-12-30:
This is the part that triggered me to file the ticket:
https://makerworld.com/en/models/94656

The teeth of the key are basically this, except it won't render:
difference() {
    tube();
    for i = [0:7] rotate() arc(...., wedge=true);
}

I ran into the exact problem on another part, too.  Looking back at that
code, my solution was to replace arc(......, wedge = true) with hull()
{arc(small radius) arc(large radius)}.   If this were my API, I would add a
new argument such as wedge2, and the result would be the hull() version,
with small radius = 0.1, or something like that.

Happy new year,
Dan

revarbat, 2023-12-31:
If I'm reading that code right, you're differencing a 2D shape from a 3D shape, which OpenSCAD doesn't allow.

dan-p3rry, 2023-12-31:
That was pseudo-code from memory, I left out a linear extrude.
Dan

revarbat, 2023-12-31:
Ah, yeah, I was reading it on a phone, but that should have been obvious to me.

So I tested some actual code that looks about like what you were describing, and came up with this:

```openscad
difference() {
    tube(od=100, wall=2, height=10, anchor=BOT);
    linear_extrude(height=2, center=true)
        for (i = [0:7])
            arc(r=60, angle=360/16, wedge=true, spin=i*360/8);
}
```

... Which fails to F6 render.  However, a minor tweak to where you execute the `linear_extrude()` can fix this:

```openscad
difference() {
    tube(od=100, wall=2, height=10, anchor=BOT);
    linear_extrude(height=2, center=true)
        for (i = [0:7])
            arc(r=60, angle=10, wedge=true, spin=i*360/8);
}
```

This renders just fine.  This is a weird OpenSCAD bug, caused by how `linear_extrude()` fails to make a manifold 3D shape because of the multi-shared edge at the origin, and CGAL (or the Manifold library) fails to handle it for the `difference()`.

I think I understand the reasoning request now.  An `ir=`/`id=` argument for `arc()` that specifies an inner radius for a wedge could be useful.

dan-p3rry, 2023-12-31:
The only difference I see in your two code sets is angle = 360/16 vs
angle=10?

And yes, adding ir/id arguments would make arc() more usable.  Thanks!
Dan

revarbat, 2023-12-31:
Argh.  That was a copy-paste error.  I've edited the comment to fix the code.  It's a matter to putting the `linear_extrude()` *inside* the for loop.

dan-p3rry, 2023-12-31:
OK, that IS interesting....
Thanks again, and Happy New Year.

adrianVmariano, 2024-01-02:
Is it better to add id/ir to arc() vs making a new ring() function?

dan-p3rry, 2024-01-02:
For me, yes.
Dan

adrianVmariano, 2024-01-02:
The question is about what makes the best, API that will make the most sense.  Why does adding ir/id to arc seem like a better option than a ring() function?

dan-p3rry, 2024-01-02:
Good point, we've diverged far from the original function of arc().  But
it's not really intuitive that a ring() function would be capable of
creating the toothed shape that I referenced.  It's your API, you have to
make the tough decisions.😀
And by the way, I use BOSL2 a LOT, so thank you for that.  My wish list for
2024 is a release version for BOSL2 so I know when to update.
Dan

adrianVmariano, 2024-01-02:
The ring function would not make a "toothed shape" but would make rings or segments of rings.  

I looked at the pic of what you're trying to make and wonder if dotted_stroke would be the way to go.

dan-p3rry, 2024-01-12:
Hi Adrian,
Did you decide on best implementation & complete?  I'm about due to upgrade
my BOSL2 code....
Thanks,
Dan

adrianVmariano, 2024-01-12:
There's been no movement on this issue.   When there is we close the issue.

adrianVmariano, 2024-01-15:
In the developer chat we're still on the fence about whether it makes sense to add a ring function/module.  What value added would it have over using stroke or offset_stroke along with the existing arc()?  Or using a difference() call, or concatenating two arcs?

dan-p3rry, 2024-01-15:
I don't know the discussions you're having, so I can only say that I have
nothing to add, and I know how to work around the problem.
Dan

adrianVmariano, 2024-02-01:
This has been added.

closed 2024-02-01 by adrianVmariano