BelfrySCAD/BOSL2 #1332: Missing a hull() for points/paths/regions that return paths/regions.

pipatron, 2023-12-20:
I can't seem to find a `hull()` that can be used for points/paths/regions. That makes it unnecessary difficult to use with the different `*_sweep()` functions and modules such as `linear_sweep()` which requires a path or region.

The candidates are [hull()](https://github.com/BelfrySCAD/BOSL2/wiki/geometry.scad#function-hull) and [hull2d_path()](https://github.com/BelfrySCAD/BOSL2/wiki/geometry.scad#function-hull2d_path). Both of them seem identical for 2D points, returning a list of _indices_ into the list of points. I don't see the difference between them, making them what I believe to be redundant?

Is there a point having two seemingly identical functions `hull(points)` and `hull2d_path(points)` when plain `hull()` can detect 2D or 3D points and work with both, or is it just an accidental development?

Since they both return _indices_ instead of a new set of points they are cumbersome to use for something like this (in increasing complexity for illustration purposes):
````
back(00) linear_sweep( circle(5) );

back(10) linear_sweep( union(circle(5),right(10,circle(5))) );

//back(20) linear_sweep( hull(circle(5),right(10,circle(5))) );
let(path=concat(circle(5),right(10,circle(5))))
  back(20) linear_sweep(select(path,hull(path)));
````

As you can see, using a `union()` is relatively clean, while hulling requires creating a new list and then extracting the points. Would it not be nice to have a `hull()` that works similar to `union()/intersection()/...` for regions? It's of course easy to write my own function but I think it would fit well for operations on regions in the library.

adrianVmariano, 2023-12-20:
It seems like what should really happen is path_idx should replace the existing path versions.  The reason for the redundancy is that the original path that we stole provided a single path function that handled all the cases through a collection of subfunctions for the different cases.  But handling all the cases isn't necessarily a great idea because it means the output can have multiple different incompatible forms, so we didn't hide the subfunctions.  

But a good path forward from where things are now is not so clear.   Leave it alone now and change hull() in BOSL2.1?

pipatron, 2023-12-21:
> But a good path forward from where things are now is not so clear. Leave it alone now and change hull() in BOSL2.1?

I'm just a single data point, but I've been happily using BOSL2 for a while now and I expected hull() to return a path or a region for 2D points, and probably a complete VNF for 3D points. The VNF version could of course keep all the original points, but that's more difficult in 2D.

adrianVmariano, 2024-01-08:
What if we made `hull_vnf()` and `hull_region()` where the former takes a 3d point list or VNF as input and returns a VNF out and the latter takes a path or region as input and returns a region?

adrianVmariano, 2024-01-14:
I've pushed changes that add `vnf_hull()` and `hull_region()` which address this.

adrianVmariano, 2024-01-14:
This has been merged.

closed 2024-01-14 by adrianVmariano