BelfrySCAD/BOSL2 #1333: [BUG] Cut half only works with certain small range of numbers

neesonqk, 2023-12-25:
**Describe the bug**
Take the example from tutorial list and change sphere to a greater number, and the cutting function is no longer taking effect.

**Code To Reproduce Bug**
```
left_half() sphere(d=70); // works
left_half() sphere(d=400); // does not work
```

**Expected behavior**
It should work with any number.

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version:  [OpenSCAD](https://www.openscad.org/) version 2021.01
- Other libraries used: None

adrianVmariano, 2023-12-25:
As described in the manual, you must set the s parameter to left_half() to give the scale.  Unfortunately it's impossible to make left_half() work on shapes of any size: the s parameter needs to be bigger than the shape but smaller than the view scale, or preview doesn't work on it.

neesonqk, 2023-12-26:
Ah, I should have read the doc more carefully. Thanks!

closed 2023-12-26 by neesonqk