BelfrySCAD/BOSL2 #1334: [BUG]

rjcarlson49, 2023-12-25:
**Describe the bug**
Customizer feature does not comply with "include" definition

**Code To Reproduce Bug**
```
/*
Include this file at the end of the Customizer variables
*/

doCrossSection = "none"; // ["none", "x", "y", "z", "-x", "-y", "-z", "polar"]
radius = 0;
phi = 0;
theta = 0;

module __Customizer_Limit__ () {}  // This actually works

_CS();

module _CS() {
    if (doCrossSection == "none") { 
        d = echo("none");
        assembly();
    } else if (doCrossSection == "polar") {
        d = echo("polar");
        crossSection(r = radius, theta = theta, phi = phi)
        assembly();
    } else { 
        d = echo(doCrossSection);
        crossSection(view = doCrossSection, r = radius)
        assembly();
    }
}
```

**Expected behavior**
When I include this file I would expect the customizer limit to be honored. It is not.

Include is described thusly: include <filename> acts as if the contents of the included file were written in the including file

**Screenshots**

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 23.12.18
- Other libraries used: crossSection is part of my own library

RAMilewski, 2023-12-25:
The Customizer is an OpenSCAD function not part of BOSL2.    

There's another way to hide variables from the Customizer.  See https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Customizer#Creating_Tabs

closed 2023-12-30 by revarbat

rjcarlson49, 2023-12-31:
My mistake. I'm so used to BOSL2 it's hard to separate it from OpenSCAD.