BelfrySCAD/BOSL2 #1335: half_of() can not handle regions as documented

pipatron, 2023-12-25:
I tried to cut a _region_ using [`half_of()`](https://github.com/BelfrySCAD/BOSL2/wiki/partitions.scad#functionmodule-half_of) from BOSL2 7714fdc but it croaked:

```
include <BOSL2/std.scad>
p = half_of(             rect(10) ,[1,0]); // Works
r = half_of(force_region(rect(10)),[1,0]); // Breaks
```

```
ERROR: Assertion 'is_path(pts, dim = undef, fast = true)' failed: "Invalid pointlist." in file lib/BOSL2/vectors.scad, line 363
TRACE: called by 'pointlist_bounds' in file lib/BOSL2/partitions.scad, line 98
TRACE: called by 'half_of' in file half_of.scad, line 3
TRACE: assignment to 'r' in file half_of.scad, line 3
```

The documentation however claims to handle regions:

>  When called as a function, you must supply a vnf, path or region in p.

…and the `p` argument:

> path, region or VNF to slice. (Function version)

I don't know if there's an error in the documentation or the code, but I would prefer to be able to slice a region directly!

adrianVmariano, 2023-12-25:
This is a bug and there's an easy fix, but I'm having trouble pushing it due to version update stuff.

adrianVmariano, 2024-01-05:
This should be fixed by the latest merge.

closed 2024-01-05 by adrianVmariano