BelfrySCAD/BOSL2 #1336: half_of() (function) mask is too small for certain 2D cases

pipatron, 2023-12-29:
I found that [half_of()](https://github.com/BelfrySCAD/BOSL2/wiki/partitions.scad#functionmodule-half_of) cut too much of my shape in certain situations, I haven't figured out exactly when yet but here is a minimal example:

```
include <BOSL2/std.scad>
shape  = pentagon(30); // Simple example path
normal = -[0.6,0.8];   // Unit normal pointing inside
point  =  [10,20];     // Point on line to slice

half = half_of(shape,normal,point);

#down(1) region(shape);
#stroke([point,point+normal*10],endcap2="arrow2");
region(half);
```
[image]

Yellow is the shape after cutting, displayed over the original shape in blue. The lower blue part should _not_ have been cut, only the top part where the normal is shown.

The normal is a unit vector, and I've found that modifying the magnitude of the normal changes the behavior. 

Looking at the code for half_of() I guess there is a problem with the way the rectangular mask is calculated, but I was not brave enough to start digging into that.

pipatron, 2023-12-29:
I wrote a slower but hopefully safer function to use in my own project, that replaces the mask calculation for 2D half_of():
```
function half_of_safe(p,v=[0,1],cp=[0,0]) =
  assert(is_vector(v,2),"Must give 2-vector")
  assert(!all_zero(v),"Vector v must be nonzero")
  let(
    // Translate the shape as if the plane is on 0->X
    a = v_theta(v),
    r = zrot(-a,move(-cp,p)),
    b = pointlist_bounds(is_region(r)?flatten(r):r)
  )
  // Keep everything X>=0
  b[0].x>0 ? r : b[1].x<0 ? [] :
    // Construct a bounding box with margins
    let(
      d = b[1]-b[0],
      e = max(d)/64,
      box = back(b[0].y-e,square([b[1].x+e,d.y+e*2]))
    )
    // Intersect the original shape with the translated mask
    intersection(p,move(cp,zrot(a,box)));
```
It's probably a little slower for complicated shapes because it translates all points in order to more easily calculate a good mask. The translated points are not used in the final intersection because that could lead to unnecessary round-off errors. Instead, the mask is translated back to the plane's reference.

adrianVmariano, 2024-01-02:
I think the original code can be fixed by setting L to the norm of the min and max bounds, basically the diagonal of the box, rather than its horizontal and vertical extents.

adrianVmariano, 2024-01-02:
```
      let(
          v=unit(v), 
          bounds = pointlist_bounds(is_region(p)?flatten(p):p),
          L = 2*max(norm(bounds[0]-cp), norm(bounds[1]-cp)),
          u = [-v.y,v.x],
          box = [cp+u*L, cp+(v+u)*L, cp+(v-u)*L, cp-u*L]
      )
      intersection(box,p)
```

Actually that's not sufficient.  The above I think fixes it.  The existing code used the un-normalized v to define the box.  You could pass a very small v and get a completely wrong result.

adrianVmariano, 2024-01-05:
This should be fixed by the latest merge.

closed 2024-01-05 by adrianVmariano