BelfrySCAD/BOSL2 #1346: Heightfield() arguments should parallel texturing arguments in cyl, linear_sweep, rotate_sweep, etc.

RAMilewski, 2024-01-14:
Heightfield uses an argument list quite different from other texturing modules.

tex_depth, tex_size, tex_reps, tex_inset would all be useful in heightfield.

"style" seems to mean different things in heightfield() vs cyl(), etc.

adrianVmariano, 2024-01-15:
It seems the heightfield() produces results rotated 90 deg from texture(), which is not great.

adrianVmariano, 2024-01-15:
Oh, actually style means exactly the same thing to heightfield as it does to texture: it's just passed through to vnf_vertex_array.

adrianVmariano, 2024-01-16:
Actually heightfield is transposing the data, not rotating.  

I think that what needs to happen is heightfield is deprecated and replaced by a module that applies a texture to the top face of a cuboid.  Not sure what it should be called.  This new module would accept the usual texturing args and would not transpose the data like heightfield does.  

There could be extra args to truncate heightfields or scale them to [0,1], I suppose.

RAMilewski, 2024-01-17:
Could it be just extending cuboid() to allow textures and specify where with "faces = "  much like rounding is specified with "edges = " ?

adrianVmariano, 2024-01-18:
How would that interoperate with chamfers and roundings?  Or would it be mutually exclusive?  

I was vaguely thinking about something that did one face of a cuboid and you could then attach() it to other objects where desired.  Another issue with multi-face support would be controlling rotation of each face.

RAMilewski, 2024-01-19:
I hadn't thought that through completely.    But if you're texturing one face, you still have the same issue don't you?

adrianVmariano, 2024-01-19:
If you're texturing one face you can rotate the texture before you put it on.   And if you're attaching a texture to several faces, you can rotate each one individually as needed.

RAMilewski, 2024-03-04:
A heightfield attached to the top of a cuboid will fail an F6 render unless bottom < 0.

RAMilewski, 2024-03-06:
Heightfields treat texture data differently than other objects:

```
include<BOSL2/std.scad>
include<data.scad>
$fn = 72;

cuboid([90,40,0.25], anchor = BOT)
    attach(TOP)
        heightfield(data, size = [90,40], bottom = -1e-12, maxz = 1);

back(25) left(30)
    linear_sweep(circle(d = 90/PI), h = 40, texture = data, tex_reps = [1,1]);

back(25) right(30)
   cyl(d = 90/PI, h = 40, texture = data, tex_reps = [1,1], anchor = BOT);
```

[image]

adrianVmariano, 2025-04-21:
Fixed by addition of textured_tile and plot3d which replace heightfield.

closed 2025-04-21 by adrianVmariano