BelfrySCAD/BOSL2 #1347: VNF textures have triangulation artifacts

adrianVmariano, 2024-01-14:


adrianVmariano, 2024-01-14:
``` 
cyl(d=10/PI, h=5, chamfer=0,
       texture=texture("hex_grid"), tex_samples=2, tex_reps=[6,3], tex_depth=.2);
```

[image]

revarbat, 2024-01-17:
This will almost certainly be a bug in the VNF slicing function.

adrianVmariano, 2024-01-17:
The artifacts arise due to application of vnf_slice to an already triangulated VNF.  The solution (verified) is to make all the texture VNFs untriangulated.  For the bricks_vnf texture

```
                [0,4,7,20], [4,8,11,7], [9,8,4,5], [4,0,1,5], [10,9,5,6],
                [20,7,6,13,12,21] ,[2,3,23,22,15,14],
                [15,19,18,14], [22,23,27,26], [16,19,15,12],[13,6,5,1],
                [26,25,21,22], [8,9,10,11],[7,11,10,6],[17,16,12,13],[22,21,12,15],[16,17,18,19],[24,25,26,27],[25,24,20,21]
```

is the correct face set.

adrianVmariano, 2024-02-02:
Two things could still be improved here.  I tried creating a single-polygon background field for cone and it runs into trouble slicing it.  If this worked it would produce the fewest artifacts.  

We could sample cone and other shapes on a grid instead of on their natural triangulation.  I believe that would produce a better result.   

We could at least for straight cylinders, not apply slicing in the Y direction, which would reduce the number of artifacts.

closed 2024-05-05 by adrianVmariano