BelfrySCAD/BOSL2 #1348: Linear_sweep() issues with textures from pngs

RAMilewski, 2024-01-14:
I appears that at least in linear_sweep() the value for tex_rot defaults to false instead of zero.  If tex_rot is not specified, the message  "boolean value for tex_rot is deprecated.  Use a numerical angle, one of 0, 90, 180, or 270." appears.

Even if tex_depth is specified, the message  "In linear_sweep() the 'tex_scale' parameter is deprecated and has been replaced by 'tex_depth'" appears.

If the source of the texture is a png processed through img2scad.py the resulting texture will be created flipped horizontally.

adrianVmariano, 2024-01-15:
Is this still true?  I thought I fixed it with the latest update.

adrianVmariano, 2024-01-15:
I think these are all fixed.  I did a test with an asymmetric texture directly specified in OpenSCAD and it came out oriented as expected, so it seems like the horizontal flip must be a img2scad problem somehow.  It appears that heightfield() produces an output which is the rotation of texture().  That seems kinda suboptimal.

closed 2024-01-15 by adrianVmariano

RAMilewski, 2024-01-15:
I can confirm that the tex_rot and tex_scale messages have been fixed.

However the horizontal flip does appear to be coming from the linear_sweep().    It's complicated a bit by the fact that img2scad.py appears to scan in column order (at least for pngs, I need to confirm it works the same way for all compatible image types).

Consider this 20x20 png:    [image]

Processing it throgh img2scad.py produces this:  [test.txt](https://github.com/BelfrySCAD/BOSL2/files/13943496/test.txt)
(The file extension has been changed from .scad to .txt because Github does not allow uploading .scad files).

```
include <BOSL2/std.scad>
include <test.scad>
$fn = 36;
zscale (1/128) color("skyblue") heightfield(test, size = [20,20], bottom = 0, maxz = 255);

right(30){
    region = rect([20,1]);
    color("lightgreen") linear_sweep(region, h = 20, texture = test, tex_depth = 1/255, tex_reps = [2,1]);
}
```
[image]

The blue heightfield requires a zrot(90) cancel the column-first processing of img2scad.py, but the green linear_extrude texture cannot be rotated to have the arrow point to the right if the bar is below the arrow.