BelfrySCAD/BOSL2 #1353: [BUG] "Mixing 2D and 3D objects is not supported" error when rendering

lijon, 2024-01-18:
**Describe the bug**
offset_stroke() only works in preview, but fails when rendering. The result is ignored.

**Code To Reproduce Bug**
```
box = square([100,100],center=true);
offset_stroke(box,closed=true);
```

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: version 2024.01.14 (git 158c5b94f)
- Other libraries used: none

adrianVmariano, 2024-01-18:
Not able to reproduce.  Your code produces:

[image]

when I render.  Using 2024.01.03.  Did you maybe combine the 2d object out of offset_stroke with a 3d object?

adrianVmariano, 2024-01-19:
You need to linear_extrude it first.

closed 2024-01-19 by adrianVmariano