BelfrySCAD/BOSL2 #1354: [BUG] path_sweep() and path_copies() create CGAL error (but ok in preview)

alastaira, 2024-01-20:
**Describe the bug**
I'm using path_sweep() along the points of a bezier path to create an organic "tentacle"-like object. I'm then adding nodules along the path using path_copies(). 
This all looks fabulous in preview, but when I attempt to render I get a CGAL union error, and "No top level geometry to render". 
If I choose slightly different values for the bezier control points, or for the size of the nodules, I can sometimes make this error disappear, but I'm unable to determine exactly what the condition that causes it is. (This is part of a larger model, where I then use zrot_copies() to create an octopus/starfish out of several of these tentacles, and I originally thought it was that that was causing the issue, but I can recreate it below without using zrot_copies).

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>
include <BOSL2/beziers.scad>

// Create points of Bezier path
pts = [[0, 0], [33.6, 9.4], [10.8, 29.5], [9.8, 36.8], [5.4, 27.9]];
bez = bezier_points(pts, [0:0.1:1]);
// Sweep along the path with a decreasing circle 
path_sweep(circle(r=5), bez, twist=0, scale=0.5);
// Add nodules along the path
path_copies(bez, n=5) sphere(r=1*(7-$idx)*.85, $fn=32);
```

**Expected behavior**
I expect the render to create the 3D object exactly as it appears in preview

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 2023.12.09, running under Windows 11.
- Thankyou for this *wonderful* library - it's really made me excited about modelling in OpenSCAD again :) x

adrianVmariano, 2024-01-20:
Your profiles for path_sweep intersect each other, which produces invalid geometry.  Oddly I'm not getting errors when I render it, but errors are expected for this sort of input as outlined in the docs.   As shown with profiles=true, width=0.1:

[image]

See how the profiles cross?   You might want to review the docs for path_sweep if this isn't making sense.   You need to split your path_sweep between the two intersecting profiles and union two separate sweeps.

closed 2024-01-20 by adrianVmariano

alastaira, 2024-01-20:
Ah, that makes perfect sense, thankyou so much for the prompt response! (and for pointing out the profiles=true option, which I'll be sure to use for debugging any issues in the future). :)