BelfrySCAD/BOSL2 #1361: [BUG] rounded_prism() fails if joint_sides == joint_bot

lijon, 2024-01-28:
**Code To Reproduce Bug**
```
rounded_prism(square([40,40]),height=20,joint_sides=3,joint_bot=3);
```
Fails with:
```
[ERROR: Assertion '(debug || bot_simple)' failed: "Roundovers interfere with each other on bottom face: either input is self intersecting or top joint length is too large" in file libraries/BOSL2/rounding.scad, line 2265]
```

Making the joint values a tiny bit different works around the problem:
```
rounded_prism(square([40,40]),height=20,joint_sides=3,joint_bot=3.001);
```

**Expected behavior**
A cube with rounded sides and bottom

**Additional context**
- OpenSCAD Version: 2024.01.14 (git 158c5b94f)

adrianVmariano, 2024-01-28:
Roundoff-error caused it to not identify the corner patch as degenerate, and then produce a bunch of repeated points.  The is_path_simple() function is supposed to ignore repeated points...but didn't.   So fixed both of these in PR.

lijon, 2024-01-31:
Let me know when I can test the fix!

adrianVmariano, 2024-01-31:
You can download my fork if you're in a hurry.  I don't know when Revar will merge it.

adrianVmariano, 2024-02-01:
Fix has been merged.

closed 2024-02-01 by adrianVmariano