BelfrySCAD/BOSL2 #1364: offset_sweep: Flag to not include top/bottom fillets in attachable() geom

lijon, 2024-01-29:
**Is your feature request related to a problem? Please describe.**
I have an `offset_sweep()` with top/bottom negative rounding (fillets), but would like to position its anchors according to the non-filleted shape. That is, I don't want the fillets to be included in the `attachable()` geometry.

**Describe the solution you'd like**
A new `atype` that ignores any negative top/bottom rounding.

**Describe alternatives you've considered**
Currently I work around it like this:
```
module my_thing(p,h,tprof,bprof,anchor,spin) {
    geom = attach_geom(region=force_region(p),h=h,cp="centroid",extent=true);
    attachable(anchor,spin,UP,geom=geom) {
        position(BOTTOM)
            offset_sweep(p,h,top=tprof,bottom=bprof,anchor=BOTTOM,spin=spin);
                children();
    }
}
```

adrianVmariano, 2024-01-31:
I implemented code to set anchors as if you had invoked linear_sweep, but I'm not sure what to call these new anchors.  For rect() we have "box" and "perim" capturing this distinction.   Here we have the "hull" and "intersect" options but then those same options applied to the linear swept version, so two new names are needed.  And "box" doesn't really seem right (unless I use the bounding box option for linear sweep).

adrianVmariano, 2024-01-31:
Hmmm.  I'm not sure where they are explained.  The "hull" and "intersection" atype settings are for use with VNFs and regions where we're trying to define anchors for an arbitrary shape.  The intersection method simply takes a ray from the origin in the direction you specify and uses as the anchor point the most distant intersection point it finds, with the direction given by the surface normal.  The "hull" method finds the line (2d) or plane (3d) that is the most distant normal to the direction and which contacts the object.  The actual anchor point is the mean of the places where the plane meets.  

So assuming I just add new atype options, I'd indeed need something like "box_intersection" and "box_hull" to cover the two new cases.   But "box" isn't very accurate---it's only treated as a box in the Z direction.

adrianVmariano, 2024-02-03:
This is implemented in the PR.  Following the pattern of rect() and trapezoid(), the default is the new behavior, where it anchors on the non-rounded object, and to get the old behavior you use "surf_hull" or "surf_intersect".

closed 2024-02-06 by revarbat