BelfrySCAD/BOSL2 #1365: [BUG] [Documentation] Broken Wiki Tutorial links

alastaira, 2024-01-30:
**Describe the bug**
Several of the links in the wiki to specific tutorials seem to be broken, pointing to a "You do not have permission to update this wiki." error (for, I'm assuming all users that are not members of the repo's organisation)

**Code To Reproduce Bug**
Click on any of the following links from the wiki tutorial page at https://github.com/BelfrySCAD/BOSL2/wiki/Tutorials
 - Ways to Make a Box (https://github.com/BelfrySCAD/BOSL2/wiki/Tutorial-Boxes)
 - Rounding Tutorial (https://github.com/BelfrySCAD/BOSL2/wiki/Tutorial-Rounding)
 - Bezier Patch Demo (https://github.com/BelfrySCAD/BOSL2/wiki/Tutorial-BezierPatch)

**Expected behavior**
Some deliciously documented tutorials, to match the excellent content found elsewhere on the wiki. Specifically, I'm hoping to find an example of how to use partition_mask with a 2D cutpath to create a custom finger joint edge between two cuboids  :)

adrianVmariano, 2024-01-31:
Someone needs to write the tutorials---the links are broken because the content doesn't yet exist.  (Though broken links aren't a great way to handle that.)   

But if you're hoping for details on partition_mask, I don't think a tutorial is presently planned that will cover that, so you might want to focus on the existing wiki docs for partitions.scad.

revarbat, 2024-04-04:
The broken links have been addressed.

adrianVmariano, 2024-04-20:
Actually "Bezier Patch Demo" still goes to a blank wiki.   Is that planned content?

closed 2024-05-05 by adrianVmariano