BelfrySCAD/BOSL2 #1371: proposal: vector_name() function

lijon, 2024-02-07:
**Is your feature request related to a problem? Please describe.**
For debugging purposes it's useful to get a string representation of a directional vector such as an anchor, orientation, face, edge, corner etc.

**Describe the solution you'd like**
I suggest adding this to the library. The name could be changed of course:

```
function vector_name(v) =
    assert(is_vector(v))
    let(
        a = ["LEFT","FRONT","BOTTOM"],
        b = ["RIGHT","BACK","TOP"],
        l = [for(i = idx(v)) if(v[i]!=0) v[i] < 0 ? a[i] : b[i]]
    )
    len(l) ? str_join(l, "+") : "CENTER";
```

**Example Code**
```
echo(vector_name(TOP+RIGHT)); // prints "RIGHT+TOP"
```

adrianVmariano, 2024-02-18:
Note that this doesn't actually check that entries are 1, and the result is arguably wrong for directions intended for use with cyl() or spheroid().   We're leaning towards skipping this, but if you want to advocate for it....

lijon, 2024-02-20:
> Note that this doesn't actually check that entries are 1, and the result is arguably wrong for directions intended for use with cyl() or spheroid(). We're leaning towards skipping this, but if you want to advocate for it....

My use case is for simple cases of (combinations of) single directional vectors, I hadn't considered cylinders and spheres. And I don't see why it's important to check for 1.0? Would it not just print the "general direction" of the vector? So `[0.1, 0, 3]` would give "RIGHT+TOP" which for me makes sense.

In any case, I have no strong opinion on having this included, I just thought it could be useful for other people as well!

adrianVmariano, 2024-02-20:
Your use case may be limited, but a library call needs to handle *all* cases intelligently.  The problem is that `[0.1,0,3]` is not in fact "RIGHT+TOP".  If you pass that to cuboid it's an error because the values are not in {-1,0,1}.   And in a context where you passed RIGHT+TOP to something more permissive, they also aren't equal.   

So ultimately the utility seems low, the complexity somewhat higher than it first appears to make this fully generic.

closed 2024-02-20 by adrianVmariano