BelfrySCAD/BOSL2 #1374: [BUG] recolor() not working inside base shape of attachable

lijon, 2024-02-10:
**Describe the bug**
`recolor()` doesn't work as expected when used inside the base shape of an attachable, but it works fine in its children.

**Code To Reproduce Bug**
```
module foo(anchor=CENTER,spin=0,orient=UP) {
    attachable(anchor,spin,orient,size=[10,10,10]) {
        cube(10,anchor=CENTER) align(TOP) recolor("green") cube(5);

        children();
    }
}
```
```
recolor("yellow")  foo() attach(RIGHT) recolor("blue") cyl(d=5,h=5);
```

**Expected behavior**
I expect a yellow cube with a smaller green cube and a blue cylinder on it.
But the inner `recolor("green")` have no effect:

[image]

Not coloring the top level object works as expected:

```
foo() attach(RIGHT) recolor("blue") cyl(d=5,h=5);
```

[image]

lijon, 2024-02-10:
Interestingly, if `recolor("yellow")` is moved into the attachable base shape, it behaves as expected:

```
module foo(anchor=CENTER,spin=0,orient=UP) {
    attachable(anchor,spin,orient,size=[10,10,10]) {
        recolor("yellow") cube(10,anchor=CENTER) align(TOP) recolor("green") cube(5);

        children();
    }
}

foo() attach(RIGHT) recolor("blue") cyl(d=5,h=5);
```

[image]

adrianVmariano, 2024-02-11:
Color is applied by attachable() to the first child, so everything in the first child gets the color.   Is there some other way it could work?

lijon, 2024-02-11:
> Color is applied by attachable() to the first child, so everything in the first child gets the color. Is there some other way it could work?

Yes this is the expected behavior, but in my first example above, recolor(green) is **not** being applied to cube(5) as expected.

adrianVmariano, 2024-02-11:
cube(5) is part of the first child to attachable().

lijon, 2024-02-11:
> cube(5) is part of the first child to attachable().

Yes, but there's a recolor(green) before it, and cube is an attachable, so my expectation is that the attachable() inside cube() would apply the color to the cube base shape.

lijon, 2024-02-11:
To clarify, I understand why this happens when looking at the code for attachable(), but I think the behaviour from a user perspective is confusing (at least it was for me). The issue happens when the base shape (first child) of an attachable() consists of another attachable. To fix it, _color() should only be applied to the inner-most non-attachable shape? But how to accomplish that, I don't know, because attachable() can't know what the first child is. However I guess it should be possible, because this works as I expect:

```
module foo(anchor=CENTER,spin=0,orient=UP) {
    attachable(anchor,spin,orient,size=[10,10,10]) {
        union() {} // dummy base shape
        union() {
            cube(10,anchor=CENTER) align(TOP) recolor("green") cube(5);
            children();
        }
    }
}
```

But I'm not sure what other surprises this might give?

adrianVmariano, 2024-02-11:
The reason your second example works as expected is that the coloring is in the second child of attachable instead of the first.  That one doesn't respond to color commands at all at the top level.  The coloring is done by the children because they are attachable---at the point where the objects are in the first child position.  

If we don't color children in the first child position....when would anything get colored?    Now I suppose in analogy with recently added (but not yet merged) use_child_tags there could be a use_child_color that would kick the color decision down one level, so to speak.  That would be equivalent to your example above with the dummy base shape.   Since it would be a special option invoked only when needed, it wouldn't prevent all the normal attachables from coloring their first child, and hence color would work.  

As a user, one could also argue that an attachable is one object, so if I ask to color it green, the whole thing should turn green, not just some parts of it.  Perhaps if you want to color different parts different colors, the right answer is to have a color argument to the module.    In the extreme case, you could make an attachable that colors all its parts and then recolor and color_this will stubbornly do nothing at the top level if the above used_child_color idea were in effect.   In other words, perhaps the "bug" here is in the documentation, not the code.

lijon, 2024-02-11:
> If we don't color children in the first child position....when would anything get colored? Now I suppose in analogy with recently added (but not yet merged) use_child_tags there could be a [...]

Yes, probably something like that.

> As a user, one could also argue that an attachable is one object, so if I ask to color it green, the whole thing should turn green, not just some parts of it. Perhaps if you want to color different [...]

My specific use case is that my project box library does a `recolor($box_inside_color)` to color all internal components to match the color of the box inside. But then, some parts are not actually box parts, but preview dummy objects (like a PCB, etc), and some of them might have multiple colors:

[image]

This OLED screen for example is an attachable that consists of the box parts (standoffs) and also the preview object. This preview object must be a *child* of the attachable, not part of its base shape, or its uses of recolor() won't work.

adrianVmariano, 2024-02-11:
So you want to make a module that has an object which is half "box" and half "not-box" and those two things are to be colored differently?   And also the objects are in a parent-child relationship so you cannot split them.

lijon, 2024-02-11:
> So you want to make a module that has an object which is half "box" and half "not-box" and those two things are to be colored differently? And also the objects are in a parent-child relationship so you cannot split them.

I think more simply phrased: I have a parent (box) with various children (parts). The box sets a default color for the children, but I want children to be able to override their color(s). The children can be any standard shape provided by BOSL2, or custom attachables.

adrianVmariano, 2024-02-11:
I don't think your simplification captures the problem, because if the parent box is a module and it has children as parts then I think everything works as-is.  The children will be able to use color_this to override.  The specific problem you're facing arises because you want to put an object AND CHILD into the first child position of the attachable and have them colored differently.  

It seems there are two solutions:  (1) the workaround you noted where you use a empty object for the first child and put your real object in the second child, or (2) add a `use_child_color=true` option to attachable(), which seems like a straight forward change, though there' s a question of how $saved_color interacts with these two cases.   Actually it turns out that it is problematic with (1), so color_this() fails in this example:

```
module foo(anchor=CENTER,spin=0,orient=UP) {
    attachable(anchor,spin,orient,size=[10,10,10]) {
       union(){};
       union(){
       cube(10,anchor=CENTER) align(TOP) color_this("green") cube(5) align(TOP+BACK) cube(1);
        children();
    }
    }
}

recolor("purple")
  cyl(d=2,h=22)
    color_this("red")foo() attach(RIGHT) recolor("blue") cyl(d=5,h=5);
```

adrianVmariano, 2024-02-12:
Hmmm.  Wondering if the names `use_child_tags` and `use_child_color` might be confusing because they seem like they could refer to children of the attachable instead of tags/colors in the attachable object itself.

adrianVmariano, 2024-02-12:
I've got use_child_color implemented in the PR, but I think maybe the name needs to be different.  Like use_object_color?   The matching use_child_tags maybe should also change.  Maybe `internal_tags=true` and `internal_color=true` for the options?  Not sure what would actually be clear without potentially being confusing.

lijon, 2024-02-12:
> I don't think your simplification captures the problem, because if the parent box is a module and it has children as parts then I think everything works as-is. The children will be able to use [...]

What I want is to group one or more 3d shapes (that are BOSL2 attachables) into a single attachable component, and be able to recolor subtrees of that component. The group of shapes could also very well use diff() and tag("remove") etc, or even tag("my_outer_remove") that works against a diff() higher up in the top level box code. (to create cutouts in the box sides). I understand such tags doesn't work in the 1st child either?

So, maybe attachable() is not meant to put other attachable(s) as the first child? And instead one should rather use the dummy union() trick and put all the grouped parts in the 2nd child? (except that this breaks color_this() as you say. Are there any other suprises going this route?)

> I've got use_child_color implemented in the PR, but I think maybe the name needs to be different. Like use_object_color? The matching use_child_tags maybe should also change. Maybe [...]

I agree that "use child ..." makes one think it's about the children (2nd child) not the base shape (1st child). `internal_` seems a better name to me!

adrianVmariano, 2024-02-12:
I do not know if there are other problems with putting the object in the second child.  I mean, if it were done systematically then nothing would ever get realized, but for "new" attachables?  I don't know.  The whole attachment system is very tricky.  Revar described it as "voodoo".  It's very hard to understand what exactly OpenSCAD is doing and how the parts interact.   I basically revamped it a while back and ended up doing a lot of trial and error coding with regards to when exactly to act on tags because I couldn't seem to reliably predict what would happen.    

It is the case that tags in the first child of attachable are not available outside of the attachable call by default.  You can still use diff and tags internally and they should work as expected, but if you want to set tags for negative space that work outside the attachable you need to either do `tag(...) attachable(...)` or you need to use the new use_child_tags option.     Note that tags can produce an exponential explosion in the number of times something gets rendered, so you want to cut that off as early as possible.  (Everything in a diff() is created 3 times.)  

Revar merged the PR that has use_child_color and use_child_tags.  These names may change, but you can try it out.

adrianVmariano, 2024-02-14:
What do you think of `override_color=false` to active this new behavior?  (And similarly, `override_tags=false`).

lijon, 2024-02-14:
I think maybe `internal_color=true` is more clear? But I'm not sure. Could one summarize the behavior as "Allow the base shape to override color set by the parent" ?

adrianVmariano, 2024-02-14:
With the "override" naming I was thinking that normally attachable() overrides any color commands (or tags) specified in the first child.  The new behavior is to turn off that override by setting it to false.  

With `internal_color=true` I guess it makes it seem like internal color commands are **always** ignored when it's false instead of only ignored if a different color is specified.   And `internal_tags=true` doesn't really seem to be great, since the basic consequence is that objects inside the first child respond to e.g. a diff() that is outside the attachable.

lijon, 2024-02-14:
This might not be a good idea, but it occured to me: why not make this new behavior the default (for both tags and colors)? The current behavior is only needed for primitives (_cube etc) so instead there could be a single `primitive=false` arg. Just thinking out loud here :)

adrianVmariano, 2024-02-15:
Well, the idea of attachables is that it's a unified object, so it's natural that it should be treated that way.  As I mentioned before, if an object is beneath $N$ calls to diff() then it will be realized by OpenSCAD $3^N$ times.  That could add up.  Having attachable() treat the object as a unified thing cuts off that explosion.  There may even be situation where you get the wrong result if you propagate into the attachable.  I don't know, but wouldn't be at all surprised.   Probably you're more likely to encounter name collisions for tag names.   Right now the only situation I know of where it's advantageous to have tags propagate is when you want to have negative space in an attachable by using a diff() above the attachable's parent.  

For color, if you use the new method then `color_this()` gives the wrong behavior, coloring only the parts of the attachable that are at the top level in the definition of the attachable object.    I hadn't noticed that before, but that could be an argument against having the feature at all.

lijon, 2024-02-16:
I see, thanks! There's many subtleties involved.
I'm closing this issue now since I believe the original "bug" has been sorted out, one has simply to learn that tags and colors behave a bit differently inside the attachable baseshape compared to its children.

closed 2024-02-16 by lijon

adrianVmariano, 2024-02-18:
I think I've fixed the problem with `color_this()` and the new behavior.  I have changed the name for this option to `keep_color=true` in the PR.  

I'm growing less certain that there is a insurmountable problem to making the new options the default, but there are complications about possible further overrides to make everything attachable (e.g. linear_extrude).  I don't think it's something to do now.