BelfrySCAD/BOSL2 #1378: [BUG] Regression in hinges

jdanders, 2024-02-15:
**Describe the bug**
Sample hinges no longer look right

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>
include <BOSL2/hinges.scad>
$fn=32;
cuboid([20,40,2])
  position(TOP+RIGHT) orient(anchor=RIGHT)
    knuckle_hinge(length=35, segs=9, offset=3, arm_height=0,
          arm_angle=90, pin_fn=8, clear_top=true);
```

**Expected behavior**
Look like the example in [hinges documentation](https://github.com/BelfrySCAD/BOSL2/wiki/hinges.scad#module-knuckle_hinge) 

**Screenshots**
[image]

**Additional context**
Git bisect shows commit 5719439b052c355a084184b0020acb8c9acd6c01 introduced the problem

adrianVmariano, 2024-02-15:
This bug is caused by a problem with anchors in offset_sweep.  The default anchor needs to be "origin" for that function/module.  Apparently it was somehow working for the hinge before when anchoring was more broadly broken.   The PR fixes this.

adrianVmariano, 2024-02-18:
This should be fixed.

closed 2024-02-18 by adrianVmariano